opentx/opentx #2915: In Firmware Simulator 2.1.3 getvalue("Cels") return a table

fred99, 2015-09-21:
After experiencing a blank telemetry script screen using 
lcd.drawNumber(130, 38, getValue("Cels"), PREC2+LEFT+SMLSIZE) 
I opened the Debug screen (F6) and I got an error message:
Script    test1 error: /SCRIPTS/TELEMETRY/test1.lua:13: bad argument #3 to 'drawNumber' (number expected, got table)

Using print(getvalue("Cels")) I got:
table: 02FD5BD8
table: 02FD5B60
table: 02FD5B60
table: 02FD5C78
table: 02FD5C78
table: 02FD5D90
table: 02FD5E08

Data are created with Telemetry simulator (F4) and CELLS = 640016 => Cels=5.00V

projectkk2glider, 2015-09-21:
The #2826 modified the return value for `Cels` You now get back the table, the number of elements corresponds to the number of battery cells, and each value holds the voltage for each cell (floating point number in Volts). How did you get your numbers, they look like hex formated. They should be floating point numbers. 

The simulation of Cels in telemetry simulator is difficult, you should provide several values for CELLS, the value is packed in specific way. Do you know how?

projectkk2glider, 2015-09-21:
@dsbeach can you post an example of Lua code that uses new Cels format?

dsbeach, 2015-09-21:
Here is the full sample code.

local cellValue = "unknown"
local cellResult = nil
local cellID = nil

local function rnd(v,d)
	if d then
	 return math.floor((v*10^d)+0.5)/(10^d)
	else
	 return math.floor(v+0.5)
	end
end

local function getTelemetryId(name)
	field = getFieldInfo(name)
	if field then
	  return field.id
	else
	  return -1
	end
end

local function init()
  cellId = getTelemetryId("Cels")
end

local function background()
  cellResult = getValue(cellId)
  if (type(cellResult) == "table") then
    cellValue = ""
    for i, v in ipairs(cellResult) do
      cellValue = cellValue .. i .. ": " .. v .. " "
    end
  else
    cellValue = "telemetry not available"
  end
end

local function run(e)
  lcd.clear()
  
  lcd.drawText(1,1,"OpenTx 2.1.x testing",0)
  
  lcd.drawText(1,11,"Cels:", 0)
  lcd.drawText(lcd.getLastPos()+2,11,cellValue,0)
  lcd.drawText(1, 56, "Mem:", 0)
  lcd.drawText(lcd.getLastPos() + 4, 56, rnd(collectgarbage("count"), 4), 0)
end

return{init=init,run=run,background=background}

Sacre100, 2015-09-21:
I found a strange behavior if you create **manualy** the Cels sensor.

Correct value on telemetry screen : 
[image]

But no cels with your script : 
[image]

If Cels is detected **automaticaly**, there is no problem.

fred99, 2015-09-22:
projectkk2glider wrote "The simulation of Cels in telemetry simulator is difficult, you should provide several values for CELLS, the value is packed in specific way. Do you know how?"
I have done a tool based on this post http://www.rcgroups.com/forums/showthread.php?t=2245978
[image]

The hearth is this code here, it creates only a lipo pack 1S or 2S max => 1 number:
// test cels.cpp : Defines the entry point for the console application.
//
#include "stdafx.h"
int _tmain(int argc, _TCHAR* argv[])
{
          _int16 cell1 = 5;
	  _int16 cell2 = 0;
	  _int16 cellNum=1;
	
	  _int16 cell1Data = cell1 * 1000 / 2;
	  _int16 cell2Data = cell2 * 1000 / 2;
	  _int32 cellData = 0;
	  _int32 firstCellNo = 0;
  
	  cellData = cell2Data & 0x0FFF;
	  cellData <<= 12;
	  cellData |= cell1Data & 0x0FFF;
	  cellData <<= 4;
	  cellData |= cellNum & 0x0F;
	  cellData <<= 4;
	  cellData |= firstCellNo & 0x0F;
	return cellData; 
}
Here I got : The program '[6304] test cels.exe: Native' has exited with code 640016 (0x9c410).

So, to use CELLS, GPS-LL, GPS-T in  the Telemetry Simulator with a LUA script,
we need to use a program to create the values and another program to decode values in the LUA script, It's a geek feature !

projectkk2glider, 2015-09-22:
>Why we don't have the simple possibility to enter until 6 cells values separated by a semicolumn in CELLS and a converter will crunch this string to create the right s.port format internally. And at the other side, getValue able to have a second optionnal parameter, an indice to get the simple value : eg getValue("Cels",4) to get the value of Cels number 4.
GPS-LL, 2 values, "longitude;latitude", GPS-T, 6 values, H;M;S;Y;M;D. Something like that..

I've added this to the #2447

fred99, 2015-09-22:
Super!
And I have forgotten, a tooltip to know what to enter like in Companion compilation options !
Better to have a built in guide.

projectkk2glider, 2015-09-23:
We can close this issue.

closed 2015-09-23 by projectkk2glider