opentx/opentx #2917: Step Range to the Adjust GV Function

voyager9, 2015-09-22:
Hello,
i would like to have a step range in the Adjust GV Function, with a Max. value.
For Example:
SF1  AileronTrimRight   Adjust GV1   STEP   (+ 1-150)    MAX  (0-150)
SF2  AileronTrimLeft     Adjust GV1   STEP   (- 1-150)    MAX  (- 0-150)

When i can choose a Step to the Adjust GV fuction it would simplify the creation of a Digital Pot. or a variable Multistep Position Switch.
The Steps are between 1-150 and the MAX 0-150 that the value cannot step over it

It would be great when you can add this function.

Best Regards, Andy

You do a great Job with the OpenTx Soft. Thumb Up

hrenz, 2015-09-22:
Well that would be fine, 
this should be same at SF increment / decrement not only with +/-1steps

free inc/dec steps  example 5 or 10 or 20 or  25

projectkk2glider, 2015-09-22:
This issue expands on #2862 (where only range was requested). I will close that one.

LapinFou, 2015-09-24:
Yeah!!
Very good idea. +:100:

bsongis, 2015-09-26:
I am afraid that we cannot add a max without a min

voyager9, 2015-10-01:
??? Why ???
The min should always be zero. Isnt it possible? Or Add a min with zero standard..... or positive as max and negative as min....

bsongis, 2016-01-02:
Refactoring explained in #3185

commit e366389c (2016-01-03): Fixes #2917
closed 2016-01-03 by bsongis