opentx/opentx #2919: companion 2.1.3 telemetry sensors simulator

wolkstein, 2015-09-22:
how can i simulate telemetry sensors (values) with taranis simulator in companion2.1.3. i noticed under 2.0x that sensors you request by lua also was filled with an value. if you open this scripts in taranis simulator with 2.1.3 all fields are empty. i know that sensors under 2.1.x need other getValu() calls. but this is the crux. it would be more easy to debug and port lua scripts to 2.1.x if all possibly telemetry sensor data (tables) could be filled by default.

/g
wolke

dsbeach, 2015-11-18:
I've started on this.  I plan to address GPS, Vario, and FLVS (lipo sensor) to start.

Are there other sensors you need?  If so, in what priority?

FYI - My goal is to be able to have Companion 'play back' sensor values from a saved log file.

bsongis, 2015-11-18:
Excellent idea, thanks for your help on this!

dsbeach, 2015-11-19:
I have a design question regarding the use of instance on the Companion Telemetry Simulator dialog box.

As currently implemented there is a single instance variable with a default value of '2'.

Option 1 - Use that instance for all simulated telemetry values

Option 2 - Add an instance text box for each simulated sensor.

I'm leaning toward option 2, but would like to identify appropriate default values and store them in a QSettings object.  Do you agree?

Status wise, the changes are looking good thanks to source code cherry picked from OpenXSensor. GPS, Vario, and Flvss data is flowing nicely!

bsongis, 2015-11-19:
@dsbeach obviously option 2) is the best :)
In the gtests directory you will find some code as well, don't hesitate to reuse it!

jtaylor2, 2015-11-19:
@dsbeach I think option 2 is better.  I added the current single instance variable as a quick fix for the D series rssi, swi, A1 and A2 being instance 25 in 2.1.  I planned on coming back and doing your option 2 (and some other improvements), but hadn't gotten to it yet.   I used a default of 2 because that is what was hard coded before my change.  Glad to see you working on it :smile:.

kilrah, 2015-11-20:
I would personally load all sensors from the simulated model with the ID/instance/name as configured, with a couple of extra sets of boxes you can fill up manually to test discovery etc. 
Then an enable checkbox per sensor, and a global simulator enable.

dsbeach, 2015-11-20:
I thought about that last night, and agree 100%. I'll investigate that approach today.

Similarly, I hope to make the process for parsing log file data like dates, and GPS positions dependent on the radio settings.

dwheeld, 2015-11-27:
Wow! A very ambitious project! I have also been working on an enhancement to the telemetry simulator (although not to the scale of yours) and am very aware of the difficulty of such a project. I wish you good luck! I will certainly keep following this thread.

pointu, 2015-12-13:
And if you can save thoses test values it should be better when testing.

dsbeach, 2015-12-13:
Closed via #3124

closed 2015-12-13 by dsbeach