opentx/opentx #2921: Simulator 2.1.3 - getValue("pilot-latitude") and getValue("pilot-latitude") return always 0

Sacre100, 2015-09-22:
In Companion Simulator 2.1.3, getValue("pilot-latitude") and getValue("pilot-latitude") return always 0 even with valid simulated GSP latitude and longitude.

GPS coord 1 : 46.515826, 6.999142 -> converted for : 27909495 -2143284163
GPS coord 2 : 46.510178, 6.992867 -> converted for : 27906106 -2143287928

Distance between point : 790m (OK in simulator).

projectkk2glider, 2015-09-23:
Not a valid field name in 2.1, see http://downloads-21.open-tx.org/firmware/lua_fields.txt

This is still missing and must be coded in 2.1.

Sacre100, 2015-09-23:
Current flight mode ("flight-mode") seems also missing.

ptsneves, 2015-10-10:
Is there a list of the missing fields so could implement them?

projectkk2glider, 2015-10-10:
Not really. You could compare the list with the 2.0 list http://downloads-20.open-tx.org/firmware/lua_fields.txt and see what is missing. You should forget about all telemetry fields that are handled natively in 2.1 (like Alt, Alt-max, etc..) and then the diff between 2.1 list should produce missing ones.

projectkk2glider, 2015-11-23:
This could be easily solved by adding additional values to the `getValue()` It currently returns latitude and longitude, we could add: pilot lat, pilot lon and distance.

mcshaz, 2015-12-05:
getValue("flight-mode") is still missing. should I open another issue?

projectkk2glider, 2015-12-05:
No, I forgot about that one.

commit 69594ba3 (2015-11-23): Fixes #2921: getValue() now also returns pilot position
PR #3097 (2015-11-24): Projectkk2glider/lua additions
Closes #3095, closes #3062, closes #2921.

**WARNING**: none of this was tested on the radio! I only did the test in the `simu` and there everything worked (I only tried `string.format()` for string library).
PR #3105 (2015-11-25): Lua addition ported from master to next
Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)

Closes  #3104

All commits from #3097 squashed into one and applied to the next branch. 

I've also added back `opent-stamp` Makefile target that I foolishly removed in 32d0e3890af7bc1c2fb23e48a7a4ddf68400d655 (I thoug [... cut]
commit 7538fcfb (2015-11-25): Git commit ID added to stamp
Fixes #2921: getValue() now also returns pilot position
Fixes #3062: Close any Lua open files when radio powers off
Fixes #3095: Lua string library added (only 6k bytes of flash used)
Re #3095: luaGetVersion() also returns version as numbers
(Ported from master 7ef790d3..4d0bd14ad and squashed into one commit)
closed 2015-12-06 by bsongis