opentx/opentx #2924: Variometer source selection OpenTX 2.1.3

pbootsma, 2015-09-23:
Taranis Plus with OpenTX 2.1.3

- The variometer source can only be selected when there is live telemetry data.
- Sometimes the variometer source selection field is not visible, not even the dashes. I had to use companion to select the VSpd and after writing to my Taranis Plus, it was visible again.

flarssen, 2015-09-24:
Only time I have seen empty fields is after deleting sensors (reference is to an empty sensor slot). But it was still possible to select another one on Tx.

bsongis, 2015-09-25:
More generally it's a problem when you remove a sensor which is used somewhere!

pbootsma, 2015-09-25:
I had not selected the vario yet, I was discovering the sensors for every model.
And I created a new model with the wizard and rebound my X4R + high precision vario , searched for sensors and the field was not there.
But I will test again.

kilrah, 2015-09-25:
Hmm there definitely seems to be a navigation issue, in the simu I can't enter edit mode on any of the vario fields if I come from "the top". If I scroll further down and come back up it's OK.

commit de4a5c58 (2015-09-25): Navigation bug (should fix #2924)
closed 2015-09-25 by kilrah