opentx/opentx #2925: Display freeze OpenTX 2.1.3

pbootsma, 2015-09-23:
I changed some telemetry lua scripts to use the new data names, but my Taranis Plus freezes the display when no telemetry data is available, for example when you do reset flight. 
Exit button and off switch do not respond, I have to pull the battery.
Flight controls remain working.

projectkk2glider, 2015-09-23:
Please provide us with your Lua scripts and eepe.

projectkk2glider, 2015-09-24:
When the freeze happens:
* do the trim keys still work (sound feedback)
* does a **long press** of EXIT key help?

pbootsma, 2015-09-24:
Trims sound works.
Long press on exit button does not work.

bsongis, 2015-09-24:
We don't have the script or it's me?

pbootsma, 2015-09-24:
local MAXPOINTS = 92
local index = 0
local altitudes = {}
local launches = {}
local lastTime = 0
local lastTimerValue = 0
local lastMaxAltitude = 0
local time
local altitude
local timer

local function init()
  for i=0,MAXPOINTS-1 do
    altitudes[i] = 0
  end
  for i=0,3 do
    launches[i] = { duration=0, maxHeight=0 }
  end
end

local function background()
  time = getTime()
  altitude = getValue("Alt")
  
  if time > lastTime + 100 then
    lastTime = time
    altitudes[index % MAXPOINTS] = altitude
    index = index+1
  end
  
  timer = model.getTimer(0)
  if timer.value < lastTimerValue then
    launches[3] = launches[2]
    launches[2] = launches[1]
    launches[1] = launches[0]
    launches[0] = { duration=lastTimerValue, maxHeight=lastMaxAltitude }
	lastMaxAltitude = 0
  end
  lastTimerValue = timer.value
  if altitude > lastMaxAltitude then
    lastMaxAltitude = altitude
  end
end

local function run(event)
  lcd.clear()
  background()
  
  -- left column
  lcd.drawTimer(1, 0, timer.value, MIDSIZE)
  local maxAltitude = getValue("Alt+")
  lcd.drawNumber(79, 0, 10*maxAltitude, MIDSIZE+PREC1)
  lcd.drawPixmap(0, 12, "/SCRIPTS/BMP/altitude-0.bmp")
  for i=0,3 do
    if launches[i].duration > 0 then
      lcd.drawTimer(10, 29+i*9, launches[i].duration, 0)
      lcd.drawNumber(70, 29+i*9, 10*launches[i].maxHeight, PREC1)
    end
  end
  
  -- middle column
  lcd.drawRectangle(80, 0, MAXPOINTS+2, 64)
  for i=1,61,2 do
    lcd.drawPoint(79, i)
    lcd.drawPoint(79+MAXPOINTS+1, i)
  end
  for i=0,MAXPOINTS-1 do
    local timeIdx
    local arrayIdx
    if index < MAXPOINTS then
      timeIdx = i
      arrayIdx = i
    else
      timeIdx = index+i
      arrayIdx = (index+i) % MAXPOINTS
    end
    local altitude = altitudes[arrayIdx]
    if altitude > 0 then
      local y = 63 - (58*altitude) / maxAltitude
      lcd.drawPoint(80+i+1, y)
      lcd.drawLine(80+i+1, y, 80+i+1, 58, SOLID, GREY_DEFAULT)
    end
    local bar = 0
    if timeIdx % 12 == 0 then
      bar = 3
    elseif timeIdx % 2 == 0 then
      bar = 2 
    end
    if bar > 0 and i<MAXPOINTS-1 then
      lcd.drawLine(80+i+1, 62-bar, 80+i+1, 62, SOLID, 0)
    end
  end
  
  -- right column
  lcd.drawPixmap(177, 36, "/SCRIPTS/BMP/altitude-1.bmp")
  lcd.drawNumber(211, 0, 10*altitude, MIDSIZE+PREC1)
  local vario = getValue("VSpd")
  if vario > 5 then
    lcd.drawFilledRectangle(176, 21, 34, 4, 0)
  else
    lcd.drawRectangle(176, 21, 34, 4)
  end
  if vario > 2 then
    lcd.drawFilledRectangle(187, 26, 23, 4, 0)
  else
    lcd.drawRectangle(187, 26, 23, 4)
  end
  if vario > 0 then
    lcd.drawFilledRectangle(196, 31, 14, 4, 0)
  else
    lcd.drawRectangle(196, 31, 14, 4)
  end
  if vario < 0 then
    lcd.drawFilledRectangle(196, 46, 14, 4, 0)
  else
    lcd.drawRectangle(196, 46, 14, 4)
  end
  if vario < -2 then
    lcd.drawFilledRectangle(187, 51, 23, 4, 0)
  else
    lcd.drawRectangle(187, 51, 23, 4)
  end
  if vario < -5 then
    lcd.drawFilledRectangle(176, 26, 34, 4, 0)
  else
    lcd.drawRectangle(176, 56, 34, 4)
  end
end

return { init=init, background=background, run=run }

pbootsma, 2015-09-24:
Eepe file follows when I have found out how to attach it.

projectkk2glider, 2015-09-24:
A problem in your Lua script:

When telemetry is reset `Alt+` becomes zero, your variable in script `maxAltitude` is zero and you use it to divide:
```lua
    local altitude = altitudes[arrayIdx]
    if altitude > 0 then
      local y = 63 - (58*altitude) / maxAltitude
      lcd.drawPoint(80+i+1, y)
      lcd.drawLine(80+i+1, y, 80+i+1, 58, SOLID, GREY_DEFAULT)
    end
```
There is also `)` missing on that line. You will have the same problem if telemetry stops, all telemetry values returned by `getValue()` are zero in that case.

But even if this happens the LCD should not freeze.

projectkk2glider, 2015-09-24:
When above happens the `lcd.drawLine(80+i+1, y, 80+i+1, 58, SOLID, GREY_DEFAULT)` the `y` is a very large negative number which causes the function `lcd_line()` to loop excessivelly long (that is  one bug). My example has:
```
Breakpoint 2, Taranis::luaLcdDrawLine (L=0x7f5324000990)
    at ../radio/src/lua_api.cpp:439

(gdb) info locals
y1 = -2147483648
x2 = 91
pat = 255
flags = 720896
x1 = 91
y2 = 58
```

Because the CPU time is used otside Lua scritp the script is not killed.

projectkk2glider, 2015-09-24:
Another comment: Division by zero is not an exception in Lua, you just get a special value `inf` as a result. The `inf` Lua value is a bit pattern of `0xfff80000` which is `-2147483648` when cast to int32_t.

projectkk2glider, 2015-09-24:
So how to fix this @bsongis ? 
* where to put the check for invalid parameters
  * in `luaLcdDrawLine()` 
  * or `lcd_line()`
* what to do if one or more parameters are invalid:
  * skip drawing
  * or modify parameter to safe value

bsongis, 2015-09-25:
I would vote for a fix in lcd_line, Other checks are done there, it would be consistent. And yes a safe value would be nice. Here y1 should be moved to 0. lcd_vlineStip does so.

PR #2929 (2015-09-25): Fixes #2925: added coordinate check in lcd_line()
I tried fixing input parameters to safe values, but the code was too complicated. Instead a simple check is done, if start or end of the line is outside the LCD the drawing is skipped altogether.

This pull probably closes #2925, but we need to verify that.
commit a38e9f6d (2015-09-25): Fixes #2925: added coordinate check in lcd_line()
commit e5546689 (2015-09-25): Merge pull request #2929 from opentx/projectkk2glider/issue_2925_lcd_line_fixes

Fixes #2925: added coordinate check in lcd_line()
closed 2015-09-25 by bsongis