opentx/opentx #2934: Screenshot,  Filenames to long, cant see at SD-Cart at Display

hrenz, 2015-09-27:
Hy,

the Screenshot Filenames are to long.
I can't see the filenames at the Sd-Card at the Display direct, not displayed.
Only if I connect to PC I can see that there are Files of Screenshot at the Sd-Card
Example:  screenshot-2015-07-27-22-44-54.bmp

Possible shorter Filename: 
Screen20150727-224454.bmp
or shorter
20150727-224454.bmp

kilrah, 2015-09-27:
The radio doesn't have the ability to read those anyway.

hrenz, 2015-09-29:
I will only see the filenames itself notthe screenshot
now I see nothing

projectkk2glider, 2015-10-08:
I vote for `Screenshot-20150727-224454.bmp` which is 30 characters (limit is 32). Minimal changes required in the code.

bsongis, 2015-10-08:
Yes, go on :)

projectkk2glider, 2015-10-09:
The same date/time generation function is used in:
* model backup (only date part)
* eeprom backup
* telemetry log (only date part)
* screen-shot

I do not want to change behaviour in all of them, so my current recommendation is to leave the date/time alone and shorten the prefix to **scr**: `scr-2015-07-27-22-44-54.bmp`. @bsongis do you agree?

bsongis, 2015-10-09:
I also think we should keep consistency in these names.
Why not screen-2015-07-27-224454.bmp?

At the beginning I thought about screen-2015-07-27-22:44:54.bmp. But the quick selection in a terminal won't work with usual xterm*charClass settings

commit a30ddb74 (2015-10-10): Fixes #2934: shortened screen-shot file name (screen-2013-01-01-123540.bmp), also affects EEPROM backup which uses same time-stamp generation function.
closed 2015-10-10 by projectkk2glider