opentx/opentx #2935: Duplication of receiver number when copying a model

Comdoc292, 2015-09-28:
Had a situation where I left a battery connect in a model I had copied in my Taranis, result was one plane took off and one in the pits ran into a fence, air born plane crashed as lost sight of it. There needs to be a warning that copying a model also copies the receiver number. If 2 receivers have the same number and both are powered, they will both respond. If you try to manually change a receiver number the software rightly will not allow this to happen, but no restriction if copying a model

kilrah, 2015-09-28:
Duplicate of #2187

closed 2015-09-28 by kilrah

griebd, 2015-09-28:
just as a clarification, which firmware do you have in your radio @Comdoc292?

Comdoc292, 2015-09-29:
I have a Taranis Plus 2.0.17

  _____  

(#2935)

<https://github.com/Comdoc292> ?

-
<https://github.com/opentx/opentx/issues/2935#issuecomment-143739164> it on
GitHub.
<https://github.com/notifications/beacon/AOLYS0vN3ZSv9S7w8GrqHJmxnKK5VKfCks5
o2TO5gaJpZM4GEp0T.gif>

griebd, 2015-09-29:
OK then, I just wanted to be sure there was no unseen bug in 2.1... For 2.0.17 there is already a fix, issue #2187. It will be available in 2.0.18.