opentx/opentx #2936: Taranis X9E SWF and SWH don't seem to recognise 3 pos on-off-on switch

cncmodeller, 2015-09-28:
Hi,
    I've fitted an On-Off-On three position switch to both SWF and SWH on my Taranis X9E and set 3 POS in the hardware menu the main screen display page only shows a 2 positions when the switch is toggled. I note that the switch port connectors header markings are in bold (like the analogue inputs) SWH rather than SW-H so is this an analogue "switch" port that I need to feed 0v - Vcc/2 -Vcc to get the three positions or is it a firmware bug??

Any thoughts gratefully received.

Barry M

projectkk2glider, 2015-10-08:
Did you configure the switch type to 3POS in radio setup menu?

projectkk2glider, 2015-10-08:
@bsongis does commit bcfbc4d04c03278a1347ae7730167c3db5d0ae89 fix this or do we have another problem?

bsongis, 2015-10-08:
I need to check, I am not sure there are 2 PINS for SF and SH! At least SH cannot be 3POS. Would you double check?

bsongis, 2015-10-08:
@projectkk2glider No this commit has nothing to do with the physical switches as they are in the beginning of the list

cncmodeller, 2015-10-08:
Both of the headers for the two switched have 3 pins on the X9E see pic opf SWH below.

Hope that helps,
Ta
BPM

[image]

bsongis, 2015-10-09:
Yes but it doesn't say that the 3 pins are connected to the CPU!

kilrah, 2015-10-09:
On the X9E ALL switches can be 3POS, they all have 2 pins that are connected all the way! Including SH

projectkk2glider, 2015-10-09:
Btw my development PCB back board has SW-H connector positioned very differently. In fact many of the connectors are placed in other positions, Did we perhaps missed some CPU pin changes here? Maybe one leg of SW switch is wired wrong in hal.h.

Other possibility is that use has a defective switch. @cncmodeller can you temporarily try to use some other (known to be working) switch in SW-H slot to see if it works.

bsongis, 2015-10-09:
@kilrah  All the schematics I have don't have 2 pins for SH

kilrah, 2015-10-09:
They must be outdated?

[image]

projectkk2glider, 2015-10-09:
We have a firmware problem, hal.h:
```
#if defined(REV9E)
  #define SWITCHES_GPIO_REG_H           GPIOF->IDR
  #define SWITCHES_GPIO_PIN_H           GPIO_Pin_1  // PF.01
#elif defined(REVPLUS)
  #define SWITCHES_GPIO_REG_H           GPIOD->IDR
  #define SWITCHES_GPIO_PIN_H           GPIO_Pin_14 // PD.14
#else
  #define SWITCHES_GPIO_REG_H           GPIOE->IDR
  #define SWITCHES_GPIO_PIN_H           GPIO_Pin_13 // PE.13
#endif
```
So SH is defined as one input switch!

projectkk2glider, 2015-10-09:
Actualy there are more errors in hal.h. At least SF, SH and SI have wrong inputs.

bsongis, 2015-10-09:
No, Adela sent me 3 or 4 versions ... I am almost sure that what we have is good and SF and SH are 2pos only, now I remember. I will try to find the last table (it is not in the schematics)

projectkk2glider, 2015-10-09:
Then we must limit the choice in hardware setup menu for SF and SH. But first we need to verify this somehow.

kilrah, 2015-10-09:
Yes, if there are 4 versions floating around we really need to ditch everything we have and ask FrSky to give us a version that 100% matches the production hardware for sure, otherwise we'll just continue to go crazy.

bsongis, 2015-10-09:
I am lost, I have too many versions sent on Skype or email, some are hand written. I really need to ask Adela to be sure things are right. But yes I remember I was lazy and didn't do what @projectkk2glider explains.

cncmodeller, 2015-10-09:
Thanks guys, 
                     Both the switches I was using work fine in all 3 positions when connected to other switch headers.

Hope that helps...
Cheers
BPM

projectkk2glider, 2015-10-09:
@cncmodeller it now looks like the SH and SF are only able to accept (physically) the 2 position switches. So if you want to use the 3 position switch, then move it to some other slot (not to big otherwise you will get another problem).

The OpenTX will have to be fixed to remove a possibility to select 3POS type for SH and SF in radio hardware setup menu.

cncmodeller, 2015-10-09:
OK thanks guys I've got a couple of two pos switches in the physical switch config I'm using so I'll move them over. I'm guessing it's the top / centre pins (as seen in the pic above) that are connected as from memory the switch registered a different state when closed across them vs open.

Thanks for your help. Great work BTW, loving the functionality. I'm finally getting all of the mode / camera and gimbal controls I've wanted for ages. Operating my quad will be relative child's play after the upgrade to the X9E and OpenTx :-)

Cheers
Barry M

projectkk2glider, 2015-10-10:
>The OpenTX will have to be fixed to remove a possibility to select 3POS type for SH and SF in radio hardware setup menu.

I will do this...

closed 2015-10-10 by bsongis

MarenB, 2015-10-25:
Hi,

I have similar issue here after I swapped SC and SF on my X9E, so SF is now 3POS, SC is 2POS. 
Adopted the settings on the hardware menue, but that does not change anything at all. I can only use two positions of SF and SC shows all three positions when I do the autodetect-thing in the mixer setting and move the 2POS-switch slowly.

kilrah, 2015-10-25:
As explaned above SF only supports a 2-pos switch, so your 3-pos obviously won't work there. 
What "3 positions" does SC show? You should have up and down but not middle.

MarenB, 2015-10-25:
Damn. Sorry. No idea why I didn't hear the bell ring - it must have nearly hit me.

I can set SC to be 2POS, but if I move it slowly, I can see it going from up to center to down. Maybe that's normal, but I just wanted to notice it. Not sure, what electrical-wise causes the center position reading.

kilrah, 2015-10-25:
Your 2-pos is a single contact that is either open or closed, right? Not closed/closed using 2 contacts?

MarenB, 2015-10-25:
It's the stock switch from SF that I moved to SC.

PR #2957 (2015-10-10): Fixes #2936: 3POS type removed from SH and SF switches (hw only suppo…
…rts 2POS type)

Closes #2936
commit 21760a39 (2015-10-10): Fixes #2936: 3POS type removed from SH and SF switches (hw only supports 2POS type)
commit e8680c32 (2015-10-10): Merge pull request #2957 from opentx/projectkk2glider/issue_2936_sh_sf_two_pos

Fixes #2936: 3POS type removed from SH and SF switches (hw only suppo…