opentx/opentx #2938: MEGA2560 the first PPM pulse in the frame longer.

tmrttmrt, 2015-09-30:
I have just noticed that the first pulse in the PPM stream is 700us long while the rest are correctly 300us long on MEGA2560. I believe this is not the correct behavior.

tmrttmrt, 2015-10-03:
Please confirm whether this is the intended behavior.

sloped-soarer, 2015-10-15:
Is the first pulse in every frame 700us long ?.
I suppose this is important if you are decoding the pulses from leading edge to leading edge rather than trailing edge to trailing edge.

tmrttmrt, 2015-10-15:
In every frame, of course. FrSky DHT transmitter module works correctly just with one polarity. 

I suspect that the behavior might be the same for 9x.. boards.

sloped-soarer, 2015-10-15:
HI Tmrttmrt, I'll see if i can re-create the issue. I have scope and logic analyser but my hardware is loosely based on the Arduino Mega2560 platform. I don't like the pulses.cpp and intend to rewrite it completely for my own hardware. Revolution not evolution : that won't be of any help to You i suspect.

P.S. Have you tried input capture of CPPM yet ?

tmrttmrt, 2015-10-15:
I have have tried no input capture and have no plan to do it.

sloped-soarer, 2015-10-15:
[image]

sloped-soarer, 2015-10-15:
It looks to me like it's adding 
#define SETUP_PULSES_DURATION 1000/*500us*/
to thr first sync pulse.

tmrttmrt, 2015-10-16:
#define SETUP_PULSES_DURATION 1000/500us/ attracted also my attention, but since I am not sure whether this is the intended behavior I did not go deeper in trying to understand the code.

sloped-soarer, 2015-10-16:
Try this in pulses_avr.cpp
Hopefully somebody could do the Git thing for Me.

// TIMER1_COMPA_vect used for PPM and DSM2=SERIAL
uint8_t g_ppmPulsePolarity = 0;
ISR(TIMER1_COMPA_vect) // 2MHz pulse generation (BLOCKING ISR)
{
	uint8_t dt = TCNT1L; // Record Timer1 latency for DEBUG stats display.

	// Call setupPulses only after "rest"" period has elapsed.
	// Must do this before toggle PORTB to keep timing accurate.
	if(IS_DSM2_SERIAL_PROTOCOL(s_current_protocol[0]) || *((uint16_t*)pulses2MHzRPtr) == 0)
	{
		if(!IS_DSM2_SERIAL_PROTOCOL(s_current_protocol[0]))
		{
			OCR1A = SETUP_PULSES_DURATION;
#if defined(CPUM2560)
			OCR1B = OCR1A;
			if( g_model.pulsePol ) TCCR1A = (TCCR1A | (1<<COM1B1)) & ~(1<<COM1B0); // Set idle level.
			else TCCR1A |= 3<<COM1B0;
			
			TCCR1C = 1<<FOC1B; // Strobe FOC1B.
			TCCR1A = (TCCR1A | (1<<COM1B0)) & ~(1<<COM1B1); // Toggle OC1B on Compare Match.
#endif
		}
    setupPulses(); // Does not sei() for setupPulsesPPM.
    heartbeat |= HEART_TIMER_PULSES;
    return;
	}

	if(s_current_protocol[0] != PROTO_NONE)
	{
#if !defined(CPUM2560)
		// Original bitbang for PPM.
		if(g_ppmPulsePolarity)
		{
			PORTB |=  (1<<OUT_B_PPM); // GCC optimisation should result in a single SBI instruction.
			g_ppmPulsePolarity = 0;
		}
		else
		{
			PORTB &= ~(1<<OUT_B_PPM);
			g_ppmPulsePolarity = 1;
		}
#else // defined(CPUM2560)
		// CPUM2560 hardware toggled PPM_out.
		if( *( (uint16_t*) pulses2MHzRPtr + 1 ) == 0) // Look one step ahead to see if we are currently the "rest" period.
			OCR1B = 0xffff; // Prevent compare match hence toggle.
		else OCR1B = *( (uint16_t*) pulses2MHzRPtr);
#endif
	}

	OCR1A = *( (uint16_t*) pulses2MHzRPtr); // Schedule next Timer1 interrupt vector (to this function).
	pulses2MHzRPtr += sizeof(uint16_t); // Non PPM protocols use uint8_t pulse buffer.

	if (dt > g_tmr1Latency_max) g_tmr1Latency_max = dt;
	if (dt < g_tmr1Latency_min) g_tmr1Latency_min = dt;
}

tmrttmrt, 2015-10-17:
At the moment I do not have time to do it. In a couple of weeks I could do it.

griebd, 2015-10-17:
@sloped-soarer git is quite simple, you should have a try... :wink:
but if you really do not want to try you can send me the fix and I will make a pull request in your name.

sloped-soarer, 2015-10-17:
Hi griebd.
I'm new to this Git stuff being a bit of an old git myself ! (do you understand My humour). I will send You the fix to review and maybe test ?. If that's okay. I have tested on Mega2560 and confirmed the Bit Bang PPM still works. Essentially one function was re-written. Is that okay ?.
Maybe then i can read "The Idiots guide to Github".

griebd, 2015-10-17:
Yes, just send it to me. I can't test as I do not have a mega2560... but I will make a branch and open a pull request with your code and people will test and approve it, or not... :wink:
Just so you know, if you decide to do it yourself, you just need to follow this:
1. fork OpenTx here on GitHub
2. use git on your computer to clone your fork
3. create a branch on your local copy and make your changes
4. push your branch to GitHub (using your local git)
5. go to your fork on GitHub and there will be an option to create a pull request back to OpenTx

sloped-soarer, 2015-10-18:
Thanks @griebd.
I took a slightly shorter route to step 5 ... Cut and pasted on Github.
Shall i create a pull request (Oh the excitement !) ?.

JivaroFAD, 2015-10-18:
Very nice to see a new M2560 enthousiat... welcome ;)

griebd, 2015-10-18:
@sloped-soarer yes, please do create! Explain your changes in the pull request and ask for others that do have a m2560 to test it.
If all goes well your pull request will end merged.
And later on, if you decide to keep contributing to the development, @bsongis will end adding you to OpenTx team :wink:

tmrttmrt, 2015-10-18:
@sloped-soarer, I am away for a week. I hope that during the next week I will be able to do some tests.

closed 2019-08-19 by kilrah