opentx/opentx #2939: Edge in 2.1.x behaves different than in 2.0.17

fraternl, 2015-09-30:
http://www.rcgroups.com/forums/showpost.php?p=32799773&postcount=1291

Someone posted this on rcgroups and I asked him to post it on github as I don't have my radio with me.

But I was just able to test this with the simulator and even use a 1-liner to make it easier to reproduce

2.1
Edge   SFup  0,0   (no release)   SAup

2.0

Edge SFup  0.0  (instant)   SAup

In 2.0 that LS can only become true if SF is up
In 2.1 that LS also becomes true if you move SA up and leave SF down

filago, 2015-10-01:
I'm the one who posted the issue on rcgroups, and the description above doesn't happen for me when I leave SF down, and is not exactly the issue I encountered.  Using the 1-liner described above:

2.1
Edge SFup 0,0 (no release) SAup

The issue I encountered is the Logical Switch becomes true if SA is moved up when SF is *already* up. I expected it to stay false because the edge event for SF did not happen with SA up.

If I move "AND SAup" to a second line, the logic should be the same, right? :

L1 = Edge SFup 0,0 (no release)
L2 = L1 AND SAup

But now L2 doesn't become true if SA is moved up when SF is already up. This matches my expectation and rev 2.0.17 single-line behavior.

projectkk2glider, 2015-10-08:
We need to first determine if this is actually a bug or intended behaviour.

filago, 2015-10-09:
I suspect there's a bug, and this should demonstrate it.  Create 3 logical switches:

L1 = Edge SAup 0,0 (no release)
L2 = Edge SAup 0,0 (no release) AND SBup
L3 = AND (L1, SBup)

L1 triggers as expected; any time SA is switched up.
L2 and L3 both add "AND SBup" to the logic, so I think they should behave identically (they don't) and they should only trigger if SA is switched up when SB is up.  L3 does behave like that.  However, L2 also triggers if SB is moved up with SA already up.  I think this is a bug, because L2 will trigger at a time when SA is not being moved.

fraternl, 2016-01-02:
@projectkk2glider have you checked if the current 2.1.x behaviour is intended?

I never seen a ticket asking for such a change and it's probably caused by some other change.

projectkk2glider, 2016-01-05:
First test on 2.0.18:
 * setup: `L1   ( Edge(SF↑, [0:(instant)]) ) AND SA↑ Duration (0.5s)`
 * when SF is kept UP and SA is moved to UP the L1 is triggered

So the 2.0.x is doing exactly the same as you describe the 2.1 is doing. No change there. Please verity your setup!

Tested the same setup on 2.1.7 and it behaves the same.
```
L1    Edge(SF↑, [0:instant]) Duration(0.7s)
L2    ( Edge(SF↑, [0:instant]) ) AND SA↑ Duration(0.7s)
L3    L1 AND SA↑
```

Perhaps there is confusion in  terminology:

| 2.0 | 2.1 | description |
|-----|------|-----|
| instant | instant | reacts when V1 goes from 0 to 1 |
| release | infinite (changed from "no release"  in 2.1.5  9ae7aea7bb493d5216bc7e6101ed4674deb3ab32 | reacts when V1 goes from 1 to 0 |

fraternl, 2016-01-05:
I didn't write it down correctly in the original post
I wrote
In 2.0 that LS can only become true if SF is up
In 2.1 that LS also becomes true if you move SA up and leave SF down

It should have been...
In 2.0 that LS can only become true if SF goes up (SA needs to be up for that too)
In 2.1 that LS can also become true if SF is up and SA goes up

In 2.0.x the first is doing the "edge stuff" the boolean in V1 can mask the whole thing (which is imho preferred and the most powerful for practical uses)

In 2.1 both booleans can trigger it.

fraternl, 2016-01-05:
2.0.18 (which wasn't released upon opening) acts different than 2.0.17 and is now the same as 2.1.x
This can cause different behaviour for people that don't expect this.

projectkk2glider, 2016-01-06:
Aha, I will check older version then. There were two changes to EDGE, see        https://github.com/opentx/opentx/releases/tag/2.0.18

projectkk2glider, 2016-01-06:
The fix in #2728 changed this behaviour in 2.0.18 (and in 2.1 also). I will look for a better solution.

projectkk2glider, 2016-01-06:
@fraternl can you test my fix? Do you need compiled firmware or Companion?

fraternl, 2016-01-06:
I only need companion. I don't have my radio with me during the week, so I need to check it with the simulator. I assume you tested it?
I can test tomorrow morning?

projectkk2glider, 2016-01-06:
Ok I will prepare you the Companion, which OS?

Some explanation of the changes:

The fix above corrects this problem:

>* setup: L1 ( Edge(SF↑, [0:(instant)]) ) AND SA↑ Duration (0.5s)
>* when SF is kept UP and SA is moved to UP the L1 is triggered

Now the L1 will not trgger. I think this is more correct and in accordance with other switch types. The additional AND switch is now processed AFTER the EDGE function was evaluated.

The behaviour is different from previous versions, but I think for the better.

fraternl, 2016-01-06:
Windows....

> The behaviour is different from previous versions, but I think for the better.

The current patch restores the behaviour of 2.0.17 and before, doesn't it?

So only 2.0.18,    2.1.3, 2.1.4, 2.1.5 and 2.1.7 are "wrong"? check?

projectkk2glider, 2016-01-06:
It should be better, because 2.0.17 had a bug #2728. Fixing that one brought up this one. Now both should be fixed. After the fix is confirmed it will be ported to 2.1.8 and 2.2.0.

lshems, 2016-01-06:
Sorry, I'm lost. In 2.0.18 I have just entered the following two setups, which do not behave similar.
[image]
is this supposed to be so, or not??

projectkk2glider, 2016-01-06:
Yes this is how 2.0.18 works and my fix (which is not in 2.0.18) will fix that.

lshems, 2016-01-06:
Ok, so I need to update for those progressing on 2.0.18.

It is fixed in 2.1.6, and broken again in 2.1.7, waiting for fix in 2.1.8 ?

I rely on this in my published programs, and are still advising only 2.0.17, but people want to move on. Since I have no free logical switches, it is a problem :), and need to know if I wait for 2.0.8

projectkk2glider, 2016-01-06:
Current state:
* 2.0.18 and >=2.1.2 work the same - ie have this AND *bug*
* 2.1.8 will have this fixed
* what is not decided is 2.0 branch:
 * we could roll back all changes in 2.1.19 and have 2.0.17 like behaviour (which will then have two bugs  #2728 and this one)
 * or make this fix to 2.0 branch also, then 2.0.8 and 2.1.19 will work the same (but different from 2.0.17)

lshems, 2016-01-06:
I think now you have mixed up some version numbers??

projectkk2glider, 2016-01-06:
I've fixed version numbers above.

lshems, 2016-01-06:
I would really like a bug fix 2.0.19, since I rely on this AND function.

If I understand it correctly, 2.0.19 would behave differently than 2.0.17, but only for the bugs. I don't see a problem in fixing bugs???

I don't like the new bug in 2.0.18, as the old one in 2.0.17 didn't bother me.

LOL

projectkk2glider, 2016-01-06:
You got it. Basically a fix for #2728 had an unforeseen side-effect: it changed the AND switch behaviour to something that is not desirable. 

When we fix it the 2.0.19 should have AND behaviour back as 2.0.17, but also have fixed the bug #2728 properly. I will prepare a testing build for Companion 2.0.19 and you can try it.

bsongis, 2016-01-06:
I am ok for a 2.0.19
@projectkk2glider may I ask you to add a non regression test?

projectkk2glider, 2016-01-06:
Sure, but the EDGE switch has so many possibilities that it will take some  time to write tests for all.. First I will compile test version of Companion for Windows for @lshems and @fraternl

bsongis, 2016-01-06:
Sure! Nothing urgent, it's because I forgot something and I don't want it happens again (unless it was not me?)

projectkk2glider, 2016-01-06:
@lshems and @fraternl The test version of Companion is here http://downloads-20.open-tx.org/test/companion_2.1.9_testing.zip

Just download and unzip in any directory. Then run companion.exe from there. It should not mess up any existing installed Companion versions, but it will use (and change if you change them) Companion 2.0.x settings.

fraternl, 2016-01-06:
That companion needs dll's with a "d" in it.
These are probably dll's for debugging?

Somehow we don't agree on how the 2.0.x releases are working.
Regarding the behaviour of this edge all versions up to 2.0.17 are the same

The anomaly I noticed upon selecting the model was never noticed by anyone and shouldn't be a problem for most. That bug didn't appear in 2.0.17. It has always been there.

2.0.18 should in my opinion be retracted to give a clear message

A 2.0.19 version could replace it with the expected behaviour of 2.0.17
With or without the bug that shortly triggers it upon selecting the model

projectkk2glider, 2016-01-06:
Are you saying that the Companion I provided above doesn't work for you? Did you extract ALL files to the same directory? It should work.

fraternl, 2016-01-06:
Companion say it needs QtGuid4.dll where QtGui4.dll is provided
It then says it needs QtXmld4.dll, where QtXml4.dll is provided
It then says it needs QtCored4.dll, where QtCore4.dll is provided
It then says it needs QtNetworkd4.dll, where QtNetwork4.dll is provided

projectkk2glider, 2016-01-06:
I will make a better package, hold on.

fraternl, 2016-01-06:
I just went through many versions:

1st    is v1 behaviour
2nd  is trigger goes true upon selecting the model

You said you are making a 2.1.9 version, but I think you mean a 2.0.19 version.
Am I correct?

2.0.13   good bad
2.0.14   good bad
2.0.15   good bad
2.0.17   good bad
2.0.18   bad good

2.1.0   good bad
2.1.1   good bad
2.1.2   bad  good
2.1.3   bad good
2.1.4   bad good
2.1.5   bad good
2.1.7  bad good
2.1.8  bad good

projectkk2glider, 2016-01-06:
Correct it is 2.0.19. The whole bunch of version mentioned here caused my naming mistake. 

Your observations about behaviour of different version is consistent with my findings.

projectkk2glider, 2016-01-06:
This should work better http://downloads-20.open-tx.org/test/companion_2.0.19_test.zip (fingers crossed) :smile:

lshems, 2016-01-06:
Phonon was missing.Just copied the exe over the 2.0.18 one in my install. I'm happy for the v1 behaviour as fraternl calls it. It works as should.

Didn't/couldn't test the selecting model thing.

:)

fraternl, 2016-01-06:
You can only test the "select model thing" with a radio which I don't have here.
I assume it's alright.... It worked fine in the 2.1.x versions

I hope 2.0.18 gets pulled, so there's less chance someone continues to fly with it.

lshems, 2016-01-06:
agree. People will get confused, and its a tricky bug. not easily noticed but likely affecting a lot of programs.

projectkk2glider, 2016-01-07:
Gtest added and tested:
* with this fix and fix from #2728 applied : PASSED
* this fix removed: FAILURE  in AND test
```
[  FAILED  ] getSwitch.edgeInstant
```

* fix #2728 also removed: AND test passes, but we get FAILURE in another test 
```
[  FAILED  ] getSwitch.edgeRelease
```

So it appears that gtests covers both failures.

projectkk2glider, 2016-01-07:
@lshems Why can't you test *selecting model thing*? 

Best plan for me now is to apply this fix to 2.0 branch and release 2.0.19. The same fix will be needed for 2.1 and 2.2 branches, since they all have this bug. @bsongis I will create a pull request now, please review it.

lshems, 2016-01-07:
I don't have a radio available that runs on 2.1.

fraternl, 2016-01-07:
Tomorrow I will load the latest nightly on my radio.
Will it be in there?

Are there 2.0.x nightlies btw?

projectkk2glider, 2016-01-07:
No nightlies for 2.0. And 2.1 does not have fix jet!

bsongis, 2016-01-07:
@projectkk2glider Perhaps you could open a pull request for master so that we have more tests coming from the nightly builds? I will review your mods now.

projectkk2glider, 2016-01-07:
The fix is now in 2.1 branch  it should be available for test in next nightly build. I will reopen this until users report back @lshems and @fraternl

fraternl, 2016-01-09:
I tested this yesterday and wrote all the results here....
Something happened or did it get deleted?

kilrah, 2016-01-09:
Are you sure you submitted? Nothing was deleted, and the "newsfeed" (that still shows things even if they were deleted) also doesn't have anything since your "tomorrow I will test..." post.

fraternl, 2016-01-09:
Well, I guess something it was something I didn't do...

I will write it down again..
Everything was good, btw...

Which doesn't help is that I'm using forkhub which doesn't show your own posts.

projectkk2glider, 2016-01-09:
@fraternl your result are here https://github.com/opentx/opentx/issues/2728#issuecomment-170034981 aren’t they?

fraternl, 2016-01-09:
@projectkk2glider 
Yes,  they are.... I got sidetracked and had several tabs open when I started looking for examples....

I checked these versions

2.1.0, 2.1.5 and today's nightly

I have this setup

L1 ( Edge(SE-, [0:instant]) ) AND SH↑ Duration(1s)
L2 Edge(SE↓, [0:0]) Duration(2s)

When I turn on the radio the switched are set as SE- and SH↑

I now noticed there are actually 3 bugs.
All of them are fixed in latest nightly

-- L1 "V1-bug" the boolean in V1 should not influence the Edge behaviour, merely masking end result (good)
-- L1 Upon selecting model a false edge value triggers (wrong)
-- L2 Upon selecting model a true edge value does NOT trigger (wrong)

2.1.1 good, bad, bad
2.1.5 bad, good, good
nightly good, good, good

I did not test the 2.0 version on the radio, but I think all of them (except 2.0.18) are like 2.1.1
2.0.18 behaves like 2.1.5

I hope a 2.0.19 gets released that will act like today's nightly.

@projectkk2glider can you confirm that this 3rd situation is now acting correctly (upon selecting a model with the edge in true it should be triggered)?

I wish #2837 was solved different.
I prefer to have everything the same....
print model is still not the same as the radio which isn't the same as Companion.

They could easily be all the same (like on the radio)

projectkk2glider, 2016-01-10:
>an you confirm that this 3rd situation is now acting correctly (upon selecting a model with the edge in true it should be triggered)?

Confirmed, when the model is loaded, the logical switches are reset, and L1 will trigger if both switches are in proper position. To try to solve this *bug* would require a mayor redesign of edge logical switch code. I think we can leave as it its.

>I hope a 2.0.19 gets released that will act like today's nightly.

Yes, the 2.0.19 will act like 2.1.8 - ie like today's nightly.

>I wish #2837 was solved different.

I have already reopened that ticket.

fraternl, 2016-01-10:
I didn't want to imply that this 3rd situation was a bug.
It does not exactly "goes" true as it was already true before selecting the model, but you can equally reason it goes true as the model did not exist...

Bertrand asked you to make a regression test for the 2 situations? A test for this 3rd situation would monitor any change of this behaviour in the future.
A change in this behaviour could indicate some unwanted behaviour, not per se this behaviour itself.

I'm glad this has been fixed.

I have a launch sequence for my DLG that's dependent on the behaviour of edge.

I once made a ticket for logging the logical switches, which should be useful to find out why some things don't work as expected.

lshems, 2016-01-10:
Launch sequence using edge for DLG. Sounds familiar, that's why I ran into
this.

Lol.

projectkk2glider, 2016-01-10:
Additional gtest added, closing.

closed 2016-01-10 by projectkk2glider

lshems, 2016-01-19:
Just bought me a spare transmitter, a version 1 taranis, upgraded to opentx 2.1.17 nightly, and guess what: edge in companion works correct, in taranis not!

fraternl, 2016-01-19:
I tested this on my radio for sure. I know this because I had to wait because I'm not with my radio during the week.

Which test doesn't work exactly?

I can only check this on Friday.
What is 2.0.19 doing?

lshems, 2016-01-19:
2.0.19 is working fine on my taranis PLUS,

i have 2.1.7 nightly on my version 1 taranis.

Is it a problem to flas back and forth between 2.0 and 2.1? Anything to take care of?

If not, I will test it tomorow the other way around.

fraternl, 2016-01-23:
@lshems  Guido, I just tested it and it is working as I expect it too.
Please test it again and describe what it is doing incorrect according to you.

Try not to take more than 1 issue in an example as this would be confusing, especially now after I thoroughly tested it and I think it was good.

This stuff is too critical to have anomalies, so please answer.

fraternl, 2016-01-24:
@lshems  could you comment?
This ticket is closed, but your comments imply that it should be opened.
Please react and either confirm that it can stay closed or reopen the ticket.

lshems, 2016-01-24:
Look, I have been testing this in my specific setup, with some tricky logic triggering launch mode.

For me, the switch sequence that I have did fine on 2.0.17, not on 2.0.18 and fine on 2.0.19. It uses the EDGE in combination with the AND switch.

The response from the system simply isn't predictable in 2.1.7 nightlies. On my Taranis Plus I've now tested it with 2.1.7 nighty, and my setup is not working reliable. On my Taranis type 1 (old) it is not functioning at all. On the companion it is reliable and functioning.

I think it has something to do with timings, and I noticed that when i check switch positions in lua, it seems Im missing a quick move here and there as well.

For me you can close this ticket, as it seems to work for your setup, and I do not have the time now to search what is the problem with my setup, especially since it is such a tricky setup. could as well be another issue.

My setup uses inputs with a switch as source, compared to an input with a max source to trigger a logical switch, and it is this Logical Switch which goes into the edge funtion with an AND function. The AND is actually working on the same logical switch again, so thats why I think timing is crucial. checking 'edge' and checking 'and' cannot be done on the same moment, so if something comes in between, you are not using the conditions from exactly the same instance.

my EPEE is to be found in my BLOG, if your interested. Part 1.

PR #3212 (2016-01-07): Projectkk2glider/issue 2939 edge problem
Closes #2939 in 2.0 branch. 

This will have to be ported to master and next also!
commit 0520127f (2016-01-07): Fixes #2939: Logical switch internal state was being reset by the AND switch, that affected its function (it looked like AND was applied before the EDGE processing, not after like it is done for other logical switch types).
commit f4378038 (2016-01-07): Fixes #2939: Logical switch internal state was being reset by the AND switch, that affected its function (it looked like AND was applied before the EDGE processing, not after like it is done for other logical switch types). (ported from 2.0 branch)
commit 0ac9998e (2016-01-10): Fixes #2939: Logical switch internal state was being reset by the AND switch, that affected its function (it looked like AND was applied before the EDGE processing, not after like it is done for other logical switch types). (ported from master branch)