opentx/opentx #2941: Companion-Display checklist not working

billmars, 2015-10-02:
In Companion, when you click the "Display checklist" checkbox it will not display the checklist when opening that model.  It is not a file access issue because if you use the "View notes" functionality it pulls the file up correctly.  Display checklist works correctly on my Taranis with the same build version so it appears to be a Companion only issue.

projectkk2glider, 2015-10-08:
Which version of Companion?

billmars, 2015-10-08:
Sorry, I thought I specified. The 2.1.3.

Bill

Total Control Panel     Login<https://asp.reflexion.net/login?domain=secure-wan.com>

To: billm@secure-wan.com<https://asp.reflexion.net/address-properties?aID=204186338&domain=secure-wan.com>

Message Score: 1        High (60): Pass
My Spam Blocking Level: Medium  Medium (75): Pass
        Low (90): Pass
Block<https://asp.reflexion.net/FooterAction?ver=3&bl-sender-address=1&hID=12688210261&domain=secure-wan.com> this sender / Block<https://asp.reflexion.net/FooterAction?ver=3&ent=1&bl-sender-address=1&hID=12688210261&domain=secure-wan.com> this sender enterprise-wide
Block<https://asp.reflexion.net/FooterAction?ver=3&bl-sender-domain=1&hID=12688210261&domain=secure-wan.com> sgmail.github.com<http://sgmail.github.com> / Block<https://asp.reflexion.net/FooterAction?ver=3&ent=1&bl-sender-domain=1&hID=12688210261&domain=secure-wan.com> sgmail.github.com<http://sgmail.github.com> enterprise-wide

This message was delivered because the content filter score did not exceed your filter level.

This e-mail and any files transmitted with it are confidential and are intended solely for the use of the individual or entity to whom they are addressed. If you have received this e-mail in error, please notify the sender immediately by e-mail reply and delete this message from your system. Please note that any views or opinions presented in this e-mail are solely those of the author and do not necessarily represent those of the company. The recipient is responsible for checking this e-mail and any attachments for the presence of viruses. The company accepts no liability for any damage caused by any virus transmitted by this e-mail.

rcbebo82, 2016-02-07:
This is not yet working in 2.1.7!

kilrah, 2016-02-08:
Invalid issue, you need to use the "simulate" button for the whole radio, not for a single model. The single model one is a shortcut that skips the radio boot process so that you can straight see the outputs without the radio being stuck on a warning (or checklist) screen.

[image]

closed 2016-02-08 by kilrah

rcbebo82, 2016-02-08:
Hi,
thx got it!

Do you know where i can find the code for reading that file in and Display
it on the lcd?

B.R.
Benjamin Boy

kilrah, 2016-02-08:
https://github.com/opentx/opentx/blob/master/radio/src/gui/Taranis/view_text.cpp