opentx/opentx #2942: Mistranslation of logical switch between OpenTx 2.0 and 2.1

RC-SOAR, 2015-10-02:
Please see attached .eepe which was created in Companion 2.0.17. 

It contains a logical switch

    L5 a < x V1=CH17 V2=100

When opened in Companion 2.1.3, the reference to CH17 is displayed  as '---' and the logical switch does not work correctly.

https://www.dropbox.com/s/5g5gr9jqwyaonfw/TransBug.eepe?dl=0

RC-SOAR, 2015-10-03:
Following further investigation, the problem appears to have been introduced in Companion 2.1.2.

projectkk2glider, 2015-10-13:
Confirmed. Conversion on the radio is OK. Only Companion is affected.

projectkk2glider, 2015-10-13:
@bsongis  I need help, the raw value for v1 (CH17) is -92. This then goes trough the sourcesConversionTable which does not have defined any transformations for negative values! So the resulting v1 value becomes 0.

bsongis, 2015-10-13:
@kilrah introduced the bug in a fix for #2787 (the cast to uint32_t is wrong)

bsongis, 2015-10-13:
Well I will say that it's wrong when importing eeproms < 217

closed 2015-10-13 by bsongis

projectkk2glider, 2015-10-13:
It works!

griebd, 2015-10-13:
just so I get a bit more of knowledge about OpenTx, the upgrade of the eeprom (when needed) is implemented differently in Companion and on the radio? I'm asking because the solution here suggests so. thanks!

bsongis, 2015-10-13:
Yes really different. Companion is able to read / write any version of the eeprom while the firmware only does conversions to the latest eeprom format, and only for his board ...

griebd, 2015-10-13:
ok, so the idea I had may be a good thing! or not, you decide... :wink:
Maybe we should implement a test/check for each board that runs the update on a generic eeprom (the more complex the better) using the radio functions and do the same through the companion function. The result from both should be bit by bit equal, right?
That way we make sure people won't have strange problems by updating one way or another. And this can point to bugs other ways unseen.

bsongis, 2015-10-13:
Yes excellent idea, do you want to give it a try?

RC-SOAR, 2015-10-14:
Thanks for fixing :)

commit 1acd23bf (2015-10-13): Fixes #2942 - Needs more tests

This bug was introduced when fixing #2787, I would like to be sure I
don't reintroduce a bug there now...