opentx/opentx #2943: Horus - Navigation design

bsongis, 2015-10-03:
As you probably know, Mike and me have received one Horus board and OpenTX port to this new board has started! The biggest change is the LCD. We now have a big 480x272 color screen. 

The navigation will be different from Taranis and 9X and we have decisions to take before starting working on the UI.

Here are the keys we have:

*At the right*
- One rotary encoder with a push button on the top
- One button at each side of the rotary encoder: Up / Down / Left / Right

*At the bottom*
- A sort of joystick with a push button on the top. What will be used for, I still have no idea
- One button at its left
- One button at its bottom

So it makes 1+4+1+2 = 8 keys for navigation. Now we have to find a solution to design something which is a intuitive as possible!

*all off topic posts will be removed, sorry for that. for those who want to know, I will open a new blog with photos and current status will be given there*

bsongis, 2015-10-03:
[image]

kilrah, 2015-10-03:
OK, an attempt...

On main view, left/right switches between screens (main view(s, but we'll probably only have one with that much screen real estate) and telemetry)
One of the left keys serves as MENU, short press = model settings, long press = radio settings as usual
UP LONG brings up stats/debug as usual
DOWN LONG brings up the popup menu (reset etc)
ENTER LONG (encoder button) brings up the model selection screen

In the menus we have a "top bar" as suggested in #1608, always visible. When on it left/right changes pages. On a given page both up/down/left/right and the encoder can be used to navigate. MENU LONG brings up the contextual menu like on the Taranis. The 2nd left key serves as EXIT, short brings back to the top bar and long goes back to main view.
Maybe a long press of left/right could jump to the previous / next pages directly, but would need to try to see if it's not too much of a pain and give unwanted page changes.

bsongis, 2015-10-03:
I do agree on everything, I would only see [Enter Long] (encoder button) instead of [Menu Long] for all contextual menus, so perhaps in the main view [Menu Long] would show available models.

*Side note: there is no EEPROM on this radio, so all models will be stored in the SD card*

kilrah, 2015-10-03:
> I do agree on everything, I would only see [Enter Long] (encoder button) instead of [Menu Long] for all contextual menus

But that would conflict with the various places where we use ENTER long for editing, no? We do use ENTER LONG for some editing stuff, and have the contextual menu on MENU long on the Taranis...

> In the Main Views we could have [Up] / [Down] working as [Right] / [Left]

Yep sure.

Sacre100, 2015-10-03:
An idea is to use the analog joystick and the two buttons on left to control the mouse similarly to the Track Point on IBM/Lenovo Thinkpad.

kilrah, 2015-10-03:
Given that it's a much less efficient way of navigating I doubt that would make any sense now, but yes, if a future feature ever happens to benefit from it that's something that could be done.

Alexander-C, 2015-10-03:
My Futaba 10c uses a little joystick to navigate round the menus, the rotary encoder is used for changing vakues and the enter key in the centre of the encoder confirms the settings. Works well.

Sacre100, 2015-10-03:
> Given that it's a much less efficient way of navigating I doubt that would make any sense now, but yes, if a future feature ever happens to benefit from it that's something that could be done.

Perhaps pointing device seems less efficient but there is studies which show that wheel control are less appreciated by end user than touch-pad, touchscreen or track-point. Moreover, with rotary encoder, end user make much more mistake than with other device and sometime crash due to these error.

MIKE-DAILEY, 2015-10-04:
I’m really looking forward to having an X12D, so I’m very pleased that the porting work as started.  I’ve only been using OpenTX since March so I’m really new at this.  But a couple of ideas from a neophyte.  
 
As you are thinking about the navigation interface, is the screen resolution high enough that you could have the upper part of the screen for all the programming functions that we see today on the Taranis (icons added) and the lower part for telemetry pages?  The advantage would be when you were in flying mode the default screen that shows model name, trim, switch settings, etc. would be view able and at the same time the telemetry section would be below it. 
 
I’ve looked at the picture of the main board and the X12D picture and it is a little hard to understand how the analog joy stuck on the left works.  If it is not usable for navigating the on the screen does it make sense to use it as a input to channels so it could be used to move X – Y axis servos for a remote camera, etc. on an aircraft.  I assume it would be a bump like input in and you would need set the servo speed slow because it does not look like it is proportional.

projectkk2glider, 2015-10-04:
>I’ve looked at the picture of the main board and the X12D picture and it is a little hard to understand how the analog joy stuck on the left works.

That joystick is proportional! Perhaps the resolution of it is not as precise as the main stick, but it should still be useable for controlling lets say a camera gimbal or something similar. I definitely want an output of that stick to be used as a input into inputs/mixers/lua scripts.

kilrah, 2015-10-04:
> My Futaba 10c uses a little joystick to navigate round the menus, the rotary encoder is used for changing values and the enter key in the centre of the encoder confirms the settings. 

That's what we'd be doing, except with the 4 directional keys instead of a joystick (hated the 10C joystick by the way, most awkward and inconvenient thing I've used on a radio, was a pain to continuously have to move your finger between 2 distant controls), and the encoder can also be used to navigate.

> does it make sense to use it as an input to channels so it could be used to move X – Y axis servos for a remote camera, etc. on an aircraft. 

I believe that's precisely what it's there for.

> Perhaps pointing device seems less efficient but there is studies which show that wheel control are less appreciated by end user than touch-pad, touchscreen or track-point.

Must not be in the RC world then, because there it's usually the exact opposite. There's a reason why pretty much every radio manufacturer now uses encoders.

> Moreover, with rotary encoder, end user make much more mistake than with other device and sometime crash due to these error.

???

MIKE-DAILEY, 2015-10-04:
More on the idea on having the telemetry screens on part of the screen all the time and the reason. I’m suggesting the idea in this thread is because it would impact the navigation that you are discussing.  

I wouldn’t envision that the format of the original screens would not change much from the way they are now; idea is there would be two images on the screen at the same time like in the image below.  What I’m not sure about is how the navigation would work.  There would need to be some way to cause the telemetry section scroll through the three potential telemetry screens.   See image below. 

Not understanding the OpenTX code I realize it might be way too many changes in the code to have both screens at the same time, but with such a large screen it would be really useful.

[image]

bsongis, 2015-10-11:
Thanks for all the comments here. The result is nice and intuitive.

closed 2015-10-11 by bsongis

hrenz, 2015-10-17:
Hy,
Does the Horus has only the external XJT HF-Modul ?
No internal XJT HF-Modul?
That would be a both!
Could anyone check this?

Webbsolution, 2016-08-16:
Hey guys - I have a X12S in my hot little hands for testing - I am eagerly awaiting the open TX release can I get a primer on the readiness in a document or link ?

kilrah, 2016-08-16:
You want to join on http://opentx.rocket.chat and make sure you're also registered with FrSky.

Webbsolution, 2016-08-16:
I am a registered NDA end user / tester with Frsky - I have registered with
Rocketchat