opentx/opentx #2944: Throttle disable/hold switch global. (SAFETY)

rguptaca, 2015-10-04:
Currently I am configuring each model individually with a throttle disable switch. For safety it would be nice to have the option to set globally once. In my case I override CH1 to -MAX. Just disabling THR stick may not be enough when differential throttle is set up in mixes. Differential THR is not common and when it is implemented usually it is a very low percent unless it is an advanced user doing aerobatics. 

Maybe there is a way to account for different channel orders such as RETA, TAER. In my case Ive decided to stick with TAER as far as possible. 

Similalrly for other global functions such as:
- volume controll (mentioned in another isuue) critical becuase the wife does not appreciate being woken up:)
- range check (which would be nice if easier to access because we are supposed to do this regularly according to MAAC here in Canada). Easily accessible on my Dx5e using the momentary switch with no safety concerns. Maybe add a condition that THR must be down for bind to activate. 
- bind (not critical but helps transition from Spekteum Dx5e)
Another way to look at this that we try and use the same switch for the same actions in all our models such as gyro, mode, lights, gear to name a few. Again, this a an option to offer. Maybe models can over ride global functions. 

I'd be happy to open individual issues if there is developer buy in. I don't to waate your time opening up a bunch of issues and have you closing them. At the very least I hope the throttle disable/hold has buy in. Again it is an option for safety.

projectkk2glider, 2015-10-08:
Are you aware of global custom functions in OpenTX 2.1.x? Can it solve your throttle and volume issues?

The range check and bind are a part of **dangerous** functions for which an issue is still open #2648

closed 2015-12-23 by kilrah

smartbird34, 2016-09-14:
I have overrides only per model. But put the playtracks in the global.  I thought I would like the override safety in global also.  The logic maybe locked to current due to error modes from receivers can change the resulting ports and signal directions.  And feeding a hardcoded at limit into the input or mix can be flipped and port moved by the following processes of mix and output.   And different receivers may take the TAER and reorder which I recently learned.  So the currently setup seems the most logical and safe.  

I was recently bit (beyond counting stitches) by an electric during binding with several assumptions wrong like other receiver manufacturer moved the throttle to another port.  I now support the disable and full -/+ overrides be in the individual chains as the receiver choice is a variable per-setup.

smartbird34, 2016-09-14:
I have my volume on my Plus on S1 in global.  But the boot-up routine has a different setting until the load of this assignment takes over.  "SF1 ON Volume .S1"  Earphone after dark work but will probably be very loud to the wearer before you are able to adjust.  They port must be expecting a lot high ohm rating than the 4 & 8 ohm typical for ear buds.  It is really clear and strong.  Perfect to wiring to the field PA at special events to fly to the music.

RCdiy, 2016-09-14:
My request is choice. Who decides what is dangerous? Anything can be dangerous.