opentx/opentx #2945: Taranis B OpenTX 2.1.4N478 - D8 protocol - Bind OK, But No Rx servo pwm out

483917, 2015-10-04:
Windows 10 pro swedish
Companion 2.1.4N478 English
Taranis B with, opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N478.bin

This morning i updated my Companion 2.1.4N468 to 2.1.4N478 and the firmware in my Taranis
from "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N468"   to
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N478"

and notised that my D8 recivers Bind OK, but i get no pwm out to servos/ESC ?

So i downgraded my "Taranis B" back to , "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N468.bin"
and all working good again ,

projectkk2glider, 2015-10-04:
These are the changes https://github.com/opentx/opentx/compare/412d8131f736228279800d751e57575d9ea1cae2...e3dee13cb3500643e367ae28e628489729db0833

But I can't see anything that could affect this.

kilrah, 2015-10-04:
Me either. Please double-check.

483917, 2015-10-04:
Ok did some more testing ,
Taranis B and D4R-II D8 protocol, X8R - s.bus D16 protocol
and 4 models with D8 and D16 protocol

with firmware
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N468.bin"
servos are moving , telemetry OK ,all OK

with firmware
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N471.bin"
Or  "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N478.bin"

 I get green light on reciver "bind" and telemetry is working, but servos do not move,
on my X8R - s.bus in D16 mode, conected  to KK2 board i can se all chanels locked to 1488us

kilrah, 2015-10-04:
Please upload your EEPROöM file somewhere so we can see your settings.

483917, 2015-10-04:
Ok here is my eeprom 
https://www.dropbox.com/s/j740jfz2ew99h8w/eeprom-2015-10-04-18-31-29.bin?dl=0

The models i tested are:
Velox300 KK2 - D16 protocol
Xbird250Clf - D8 protocol
Walkera 36B - D8 protocol

chrisl42, 2015-10-09:
Just read this, same behavior for me. 469 is still working 472 is not. Unfortunately i did not download the versions in between. So it happened probably in 470 or 471.
My setting for download: ﻿opentx-taranis-haptic-timer3-lua-sqt5font-de-2.1.4N4xx.

projectkk2glider, 2015-10-09:
Diff between 469 and 472 is https://github.com/opentx/opentx/compare/6fe12ba6251ac05ed8f38548b487f91645ac0c75...bb1d368c039ecf5208c0e0f119bcc12c88cef7c3

projectkk2glider, 2015-10-09:
I will try to reproduce this on my radio over the weekend. I have Taranis revB and D8 receiver.

@483917 and @chrisl42 when you have this issue, what does the channel monitor display. Do the channel values move or not?

chrisl42, 2015-10-09:
Just tried it, channel monitor works like normal, seems that the signal is cut off later in the chain.
Also tried a self-compiled version of the code of today, that works fine concerning this issue (compiled on Mac with 
MAKE ARITHMETIC_OVERFLOW_CHECK=YES AUDIO=YES AUTOSOURCE=YES AUTOSWITCH=YES BATTGRAPH=YES BOLD=YES CORRECT_NEGATIVE_VALUES=YES CURVES=YES DBLKEYS=YES DSM2=YES FONT=SQT5 GVARS=YES HAPTIC=YES HELI=YES NAVIGATION=YES PCB=TARANIS PPM_CENTER_ADJUSTABLE=YES PWM_BACKLIGHT=YES VOICE=YES)
483 downloaded by companion of this morning does not work :-((

483917, 2015-10-09:
@projectkk2glider , My channel monitor display on Taranis is working
channel values are moving mormal , but servo pwm on Rx is locked when using,
opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N471.bin or newer

projectkk2glider, 2015-10-09:
Thanks for your reports. It then seems to be the compilation server issue. @bsongis and @kilrah did you change anything in that time that could affect this?

bsongis, 2015-10-09:
I didn't do any change on the compilation server. Nothing obvious on the code side. We need to start a dichotomia ...

projectkk2glider, 2015-10-10:
Did a test:
* Taranis B01
* loaded opentx-taranis-lua-en-2.1.4N484.bin (downloaded from the server with Companion)
* I was on 2.0.17 previously, models converted
* used a model with D6R receiver

It works. I will next try with more fw options (except haptic which I don’t have).

projectkk2glider, 2015-10-10:
Test with opentx-taranis-timer3-lua-mixersmon-sqt5font-en-2.1.4N484.bin **also works**. That leaves out just a haptic option (which both users above have). I have to first check if enabling it won't hurt my radio.

projectkk2glider, 2015-10-10:
Test with opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N484.bin **does NOT work**. So the breaking point is the haptic option.

projectkk2glider, 2015-10-10:
Self compiled (same commit as N484) with the same options as compilation server - **does NOT work**.

Using 
```
make PCB=TARANIS FLIGHT_MODES=YES HELI=YES TEMPLATES=YES PPM_UNIT=PERCENT_PREC1 LUA=YES HAPTIC=YES GVARS=YES FONT=SQT5 OVERRIDE_CHANNEL_FUNCTION=YES MIXERS_MONITOR=YES TARANIS_INTERNAL_PPM=NO SHUTDOWN_CONFIRMATION=NO TIMERS=3 SUPPORT_D16_EU_ONLY=NO OPENTX_VERSION_SUFFIX="N484" TRANSLATIONS=EN
```
And gcc `4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]`

Will try other compiler versions next.

projectkk2glider, 2015-10-10:
Disabling haptic and doing self build - **works**.

Using gcc `4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]` did not help.

Next I will disable other options one by one to see when/if it starts working.

projectkk2glider, 2015-10-10:
Removing other options (sq5font, mixersmon and timer3) had no effect. 

Using `git bisect` I determined that commit 60f2535212b04d06202e679f11299c16298d03a3 broke this (remember only when **haptic** option is selected).
```
opentx $ git bisect log
# bad: [bb1d368c039ecf5208c0e0f119bcc12c88cef7c3] Merge pull request #2933 from hrenz/patch-11
# good: [6fe12ba6251ac05ed8f38548b487f91645ac0c75] Cosmetics
git bisect start 'bb1d368c039ecf5208c0e0f119bcc12c88cef7c3' '6fe12ba6251ac05ed8f38548b487f91645ac0c75'
# good: [1d6adcc25a34c8f02b6406b215f58e366ff1bd1a] Fixes #2870
git bisect good 1d6adcc25a34c8f02b6406b215f58e366ff1bd1a
# good: [3f6e1d0b3a36857371b67e31e57a4c8c5a6cba8d] Lua error display function reused
git bisect good 3f6e1d0b3a36857371b67e31e57a4c8c5a6cba8d
# bad: [7ec4f508eec571e8331d25c50fa2596ce099384d] Merge pull request #2930 from opentx/bsongis/issue1914_by_dsbeach
git bisect bad 7ec4f508eec571e8331d25c50fa2596ce099384d
# bad: [60f2535212b04d06202e679f11299c16298d03a3] Fixes #1914
git bisect bad 60f2535212b04d06202e679f11299c16298d03a3
# first bad commit: [60f2535212b04d06202e679f11299c16298d03a3] Fixes #1914
```

bsongis, 2015-10-10:
I don't see anything wrong in this commit (and nothing related to Haptic as well)!
Would we have a buffer overflow somewhere?
What if you comment out HAPTIC_HEARTBEAT() in the 5ms interrupt?

projectkk2glider, 2015-10-11:
I've tried several things, none of them helped:
* commented out HAPTIC_HEARTBEAT() - not working 
* commented haptic driver function contents (init, on, off) - not working 
* return at the first line of every public method of haptic class - not working 

Also when the radio is not working (this bug) I get the occasional  RSSI low warning even if the signal should be perfectly OK. I is a side effect of this bug.

Then I compared memory map before and after the commit. I also compared the map when haptic is on and off. The common thing to all this comparisons is that the position of variables in RAM changes **a lot**.

So I am pretty sure we have some buffer overflow somewhere else that is causing this. Now finding this without the debugger will be tricky.

My first obvious suspect was `s_current_protocol` which was at the beginning of RAM when not working and some 20 bytes lower when working. But adding a guard buffer before it (that was a difficult task) did not help. So it is something else.

@bsongis do you have a Taranis with a SWD port ready and open? The debugger would surely reveal what is happening.

projectkk2glider, 2015-10-12:
The bug was actually found by @bsongis after two evening sessions of change/compile/trace/test and chatting between me and Bertrand.

closed 2015-10-12 by projectkk2glider

bsongis, 2015-10-12:
Thanks Damjan, but the bug was also mine ... It's there since 2.1.0, depending on compilation options!

chrisl42, 2015-10-13:
I am not that experienced with it, but it seems that it was a kind of tricky bug. Thanks a lot for fixing it !

projectkk2glider, 2015-10-13:
@chrisl42 did you test it yet? Does it work?

483917, 2015-10-13:
I will test as soon i get home ,and report back later to day

483917, 2015-10-13:
Ok i tested firmware "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N488.bin"
on all my models with D8 - D16 protocol and everything worked perfectly,
well done guys :+1:

chrisl42, 2015-10-13:
Oh sorry, didn't mention that, works for me too, thanks a lot !

commit cb5bb830 (2015-10-12): Fixes #2945: fixed bad alignment of modulePulsesData