opentx/opentx #2949: Corrupt LCD when splash disabled plus long boot time.

CommanderBatFink, 2015-10-07:
Firmware: opentx-9xrpro
Version: 2.1.3
When the "splash" screen is disabled, the LCD displays what appears to be a corrupt spatter of pixels on the LCD.   This is often, but not always, accompanied by a very long boot time before "normal" transmitter function can be assumed.
[image]

projectkk2glider, 2015-10-08:
Could be related to #2852

bsongis, 2015-10-08:
Yes, the lcd is not cleared and it is the same problem.

closed 2015-10-08 by bsongis

pfeerick, 2016-02-28:
I don't think this is a closed issue. #2852 Exaggerated this issue due to long boot times - either due to slow SD cards, or file system corruption of some description... but even it the SD card is removed, you get this screen corruption when the SD card is removed (even on the most recent v2.1.7), just not for as long a period.