opentx/opentx #2950: [OpenTX 2.1.3][Bug][9XR-Pro] Battery gauge bar shows only 1/2 battery when 100% full

biasedlogic, 2015-10-07:
If using the graphical battery gauge bar option on a 9XR Pro, the same number of bars is displayed for a full battery as on a Taranis, but Taranis has a smaller battery symbol. So the "full" charge does not "fill" the battery icon - it looks half empty.

projectkk2glider, 2015-10-24:
Image from the other issue: http://fs5.directupload.net/images/151024/zrj4qnjy.png

bsongis, 2015-10-25:
Done, needs tests

closed 2015-10-25 by bsongis

biasedlogic, 2015-10-25:
Works better, but displays always at minimum one bar of battery - even if the battery voltage is below the lower limit of the voltage range.

I'd prefer the battery symbol to be 1px narrower, so that when the battery is full there's not so much white space to the right of the "bar" - but it's a matter of how it looks like and personal preference

bsongis, 2015-10-25:
At the right of the bar is the flight mode

biasedlogic, 2015-10-25:
Sorry, I wasn't precise again ;-) 
The bargraph is displayed within a black rectangle, on a white field that is shaped like a battery. The very first bar is one pixel away from the left edge (the negative pole) of the battery. The bars fill up the battery symbol, but when "full" the rightmost bar is three pixels away from the right edge (the positive pole) of the battery symbol, giving the impression "90%full". It either needs a bar more or a shorter field. 
Regards, 
M.

PR #2909 (2015-09-20): Fixes #2950: Mavlink build option was not working
Closes #2950

Please check my changes in myeeprom.h carefully!
commit 18ffe9f5 (2015-09-20): Merge pull request #2909 from opentx/projectkk2glider/issue_2905_mavlink_broken

Fixes #2950: Mavlink build option was not working
commit 407289e1 (2015-09-20): Fixes #2950: Mavlink build option was not working