opentx/opentx #2951: [OpenTX 2.1.3][Bug][9XR-Pro] Telemetry bar screen config - selection of variable broken

biasedlogic, 2015-10-07:
When configuring telemetry screen with bars (as opposed to the textual values) on a 9XR Pro with OpenTX 2.1.3, when selecting variables directly (as opposed to pressing MENU and then scrolling through them) with left-right keys the radio starts going through the variables very fast and then goes on beeping/vibrating as if one held the key pressed indefinitely.

biasedlogic, 2015-10-08:
Video of the issue:
https://youtu.be/KDp_IMBKhZU

bsongis, 2015-10-12:
Would you check it's fixed in the next nightly builds? I only tested on the simulator

closed 2015-10-12 by bsongis

biasedlogic, 2015-10-13:
I will try checking it, I'm on Ersky9x now, so it will take a second or two to backup the radio and give it a go.

biasedlogic, 2015-10-21:
Tested, the direct selection is completely blocked - was it the expected solution?
(one has now to press MENU first, before one can select a variable for a bargraph display)

regards,
Marek

bsongis, 2015-10-21:
Yes, at least there is no strange effect! I doubt I did anything different in previous releases, am I wrong?

biasedlogic, 2015-10-21:
Hi, 
I don't know about previous releases, I actually only tried 2.1.3 as I couldn't get my custom telemetry to work under Ersky9x. Now that the bug in Ersky9x is fixed I'm back in that camp, since the 'hang on boot' bug is much more annoying :-)

commit 7297669c (2015-10-12): Fixes #2951