opentx/opentx #2953: Ambiguity in ZSTR_TEMP referring to both TEMP1_ID and TEMP2_ID

ptsneves, 2015-10-09:
This ambiguity causes problems with the scripts' `getValue()`. 

A simple rename is unadvised because the user doesn't know which of the "discovered" sensors represents the intended data.

Also there is an inconsistency in frsky_sport.cpp where sport sensors for `ZSTR_TMP`

 is valid for the range `T1_FIRST_ID, T2_LAST_ID` when the `T1_FIRST_ID, T1_LAST_ID` and  `T1_FIRST_ID, T1_LAST_ID` already exist.

I have not created a pull request because this patch is very small and applicable to 2.1.3 tag and the master and next branch are already too far ahead and with merge conflicts in frsky_d.cpp. 

If anybody can guide me where the changes of frsky_d.cpp are applicable in the `next` or `master` branch let me know and I will make a pull request

```
diff --git a/radio/src/telemetry/frsky_d.cpp b/radio/src/telemetry/frsky_d.cpp
index 49e3e9c..5abcb8e 100644
--- a/radio/src/telemetry/frsky_d.cpp
+++ b/radio/src/telemetry/frsky_d.cpp
@@ -513,8 +513,8 @@ const FrSkyDSensor frskyDSensors[] = {
   { D_A2_ID, ZSTR_A2, UNIT_VOLTS, 1 },
   { RPM_ID, ZSTR_RPM, UNIT_RPMS, 0 },
   { FUEL_ID, ZSTR_FUEL, UNIT_PERCENT, 0 },
-  { TEMP1_ID, ZSTR_TEMP, UNIT_CELSIUS, 0 },
-  { TEMP2_ID, ZSTR_TEMP, UNIT_CELSIUS, 0 },
+  { TEMP1_ID, ZSTR_TEMP1, UNIT_CELSIUS, 0 },
+  { TEMP2_ID, ZSTR_TEMP2, UNIT_CELSIUS, 0 },
   { CURRENT_ID, ZSTR_CURR, UNIT_AMPS, 1 },
   { ACCEL_X_ID, ZSTR_ACCX, UNIT_G, 3 },
   { ACCEL_Y_ID, ZSTR_ACCY, UNIT_G, 3 },
diff --git a/radio/src/telemetry/frsky_sport.cpp b/radio/src/telemetry/frsky_sport.cpp
index b2f6930..1df75ad 100644
--- a/radio/src/telemetry/frsky_sport.cpp
+++ b/radio/src/telemetry/frsky_sport.cpp
@@ -64,7 +64,8 @@ const FrSkySportSensor sportSensors[] = {
   { A3_FIRST_ID, A3_LAST_ID, ZSTR_A3, UNIT_VOLTS, 2 },
   { A4_FIRST_ID, A4_LAST_ID, ZSTR_A4, UNIT_VOLTS, 2 },  
   { BATT_ID, BATT_ID, ZSTR_BATT, UNIT_VOLTS, 1 },
-  { T1_FIRST_ID, T2_LAST_ID, ZSTR_TEMP, UNIT_CELSIUS, 0 },
+  { T1_FIRST_ID, T1_LAST_ID, ZSTR_TEMP1, UNIT_CELSIUS, 0 },
+	{ T2_FIRST_ID, T2_LAST_ID, ZSTR_TEMP2, UNIT_CELSIUS, 0 },
   { RPM_FIRST_ID, RPM_LAST_ID, ZSTR_RPM, UNIT_RPMS, 0 },
   { FUEL_FIRST_ID, FUEL_LAST_ID, ZSTR_FUEL, UNIT_PERCENT, 0 },
   { ALT_FIRST_ID, ALT_LAST_ID, ZSTR_ALT, UNIT_METERS, 2 },
diff --git a/radio/src/translations/cz.h.txt b/radio/src/translations/cz.h.txt
index a9f4470..df09b85 100644
--- a/radio/src/translations/cz.h.txt
+++ b/radio/src/translations/cz.h.txt
@@ -1107,7 +1107,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/de.h.txt b/radio/src/translations/de.h.txt
index d9db63a..69d71d0 100644
--- a/radio/src/translations/de.h.txt
+++ b/radio/src/translations/de.h.txt
@@ -1111,7 +1111,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/en.h.txt b/radio/src/translations/en.h.txt
index 4cafc0e..3ae867e 100644
--- a/radio/src/translations/en.h.txt
+++ b/radio/src/translations/en.h.txt
@@ -1106,7 +1106,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/es.h.txt b/radio/src/translations/es.h.txt
index 98d124f..87a78ea 100644
--- a/radio/src/translations/es.h.txt
+++ b/radio/src/translations/es.h.txt
@@ -1061,7 +1061,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/fi.h.txt b/radio/src/translations/fi.h.txt
index d028f90..4ca04a4 100644
--- a/radio/src/translations/fi.h.txt
+++ b/radio/src/translations/fi.h.txt
@@ -1061,7 +1061,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/fr.h.txt b/radio/src/translations/fr.h.txt
index 01c8db1..4514872 100644
--- a/radio/src/translations/fr.h.txt
+++ b/radio/src/translations/fr.h.txt
@@ -1098,7 +1098,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "BtRx"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/it.h.txt b/radio/src/translations/it.h.txt
index 8be1715..97abf64 100644
--- a/radio/src/translations/it.h.txt
+++ b/radio/src/translations/it.h.txt
@@ -1109,7 +1109,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/nl.h.txt b/radio/src/translations/nl.h.txt
index 757c0fa..4388553 100755
--- a/radio/src/translations/nl.h.txt
+++ b/radio/src/translations/nl.h.txt
@@ -1129,7 +1129,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/pl.h.txt b/radio/src/translations/pl.h.txt
index 1e1bcf6..458e5a6 100644
--- a/radio/src/translations/pl.h.txt
+++ b/radio/src/translations/pl.h.txt
@@ -1110,7 +1110,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/pt.h.txt b/radio/src/translations/pt.h.txt
index 9b4f9bb..b25dcb3 100644
--- a/radio/src/translations/pt.h.txt
+++ b/radio/src/translations/pt.h.txt
@@ -1061,7 +1061,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
diff --git a/radio/src/translations/se.h.txt b/radio/src/translations/se.h.txt
index 84065e2..cb7ec1a 100644
--- a/radio/src/translations/se.h.txt
+++ b/radio/src/translations/se.h.txt
@@ -1061,7 +1061,8 @@
 #define ZSTR_A4                "A4"
 #define ZSTR_BATT              "RxBt"
 #define ZSTR_ALT               "Alt"
-#define ZSTR_TEMP              "Temp"
+#define ZSTR_TEMP1             "Tmp1"
+#define ZSTR_TEMP2             "Tmp2"
 #define ZSTR_RPM               "RPM"
 #define ZSTR_FUEL              "Fuel"
 #define ZSTR_VSPD              "VSpd"
```

kilrah, 2015-10-09:
Hmm that was intentional based on the assumption that users would always rename them to something that makes sense with regard to what the sensor is used to measure.

ptsneves, 2015-10-09:
But the thing is it assumes users know what they are doing, which when we are talking about Lua scripts is not guaranteed.

My proposal still allows for your assumption, but provides an unambiguous default name. I think the same assumption you make could be done in A1 or A2 or A3 but it isn't reflected in the code, therefore I think it looks inconsistent.

kilrah, 2015-10-09:
OK, why not.

Please provide a pull request based on current master then, and we'll merge :)

ptsneves, 2015-10-09:
The pull request was submitted, compiled and live tested on a Taranis Plus with the master branch as requested.

Best regards
Paulo Neves

closed 2015-10-10 by kilrah