opentx/opentx #2954: more GVARs

schubi74, 2015-10-09:
Hello,
is it possible to implement more GVARs in openTX? I try to avoid entering numbers whenever I need more than one time. That means I use GVARs for EVERYTHING, like e.g. brake compensation for a v-tail (2 channels affected). This means that already for a 4-servo wing I use 7 GVARs and I could easily use more. Everything is so oversized in openTX (32 channels and soon 64 logical switches), so why do the GVARs remain so limited?
16 GVARs would be helpful, but more is always better ;-)
Regards
schubi74

kilrah, 2015-10-09:
Can't you move a good part of that as numbers in inputs? That's what they're for.

schubi74, 2015-10-09:
Hello kilrah,
could you please explain this approach a bit more in detail? Maybe with a short example?

kilrah, 2015-10-09:
It's a bit hard without an example of what you're trying to do. Might want to post one.

But the whole point of inputs is that you can create one for a given control, define one (or more) weight(s) for it here, and then use that input in multiple mixes so that should you want to adjust the weight of the control later you only have to do it once in the input.

projectkk2glider, 2015-10-09:
This will be hard to change.  You must remember that we have 9 GVARS in 7 flight modes, which equates to `9 * 7 = 63` values that need to be stored for each model.

Use some other method as Kilrah suggested.

schubi74, 2015-10-09:
I understand that I can define an input which only delivers a number via MAX and a weight. But how can it be applied to a mixer? In the "weight"-field of the mixer I can only put another number (no connection with the input) or a GVAR. Thats why I used them, even if I don't need flight-mode specific values.
If there's a trick, please explain it to me and I can close the issue.

kilrah, 2015-10-09:
> In the "weight"-field of the mixer I can only put another number

Yes, but that number would usually not need to be changed, so not much point in using a GVAR.

schubi74, 2015-10-09:
Yes, but for my example (elevator compensation for brakes on a v-tail plane) you would have to enter the number TWICE and you always have to take care to keep them SYNCED. This approach is prone to errors during the sync. 

What I would need is a value that can be applied to a mixer weight, but which doesn't need to be flight phase dependent. If you could add this feature I admit that the need for GVARs would be drastically reduced. In fact, it would just be necessary to add the values of inputs to the "list of values" for the mixer weight. Has this feature already been considered?

kilrah, 2015-10-10:
Again you don't show your settings so I can't be sure what I say applies to your example... But to me you can create a "Comp" input with a weight you adjust, and that you mix at a fixed 100% on both channels in the mixer. 
Instead of setting like 13% on both mixes' weight, you set your 13% once on the input.

schubi74, 2015-10-10:
I just invested a lot of time in preparing a demo setting, just to find out it cannot be uploaded to github :-(

But while I did it, I understood what you wanted me to understand. I had the input setting from 0 to 100% and then adapted the mixer weight. You want me to reverse that, to make the input from 0 to 13% and a mixer weight of 100% which after multiplication results in 13% for both cases. Then I can change my setting in the input setup for both channels.

I will need to let this sink in. Thank you so far.

Shall I close the issue?

closed 2015-10-10 by projectkk2glider