opentx/opentx #2956: OpnTxCompanion draw.channel attributes

rdeanchurch, 2015-10-10:
[image]
[image]

A small inconsistency in DrawChannel.
Attribute does not carry over to units in OpenTxCompanion 2.0.17.

kilrah, 2015-10-10:
Is there a reason for mentioning companion specifically (i.e. it works fine on the radio?)

rdeanchurch, 2015-10-10:
Mentioned on Companion only because that was the only place I tested so far.
TBDone

projectkk2glider, 2015-10-10:
The same issue is probably also present in 2.1 and should be fixed there too.

projectkk2glider, 2015-10-18:
Now that I started to port this to the 2.1 branch I realized I do not like this change!

The fact is that the function `putsTelemetryValue()` is also used from the OpenTX firmware many times and for example when editing Logical switch comparison value the correct behaviour is to have only a number part of the value blinking. 

My change in #2977 changes that and the unit also blinks which could suggest to the user that the unit can be changed.

So my vote is to revert this change and leave it as it is. If somebody wants a special formatting for their Lua script, they can do it from the Lua code. But not all changes from #2977 are bad, the change in case for Voltage unit is ok in my opinion. @bsongis and @kilrah I need your opinion.

bsongis, 2015-10-18:
Right the unit should not blink!

projectkk2glider, 2015-10-18:
I will revert the offending commits...

projectkk2glider, 2015-10-18:
Done. 

@rdeanchurch you might not like the result of this issue, because in the end nothing has been done to address your request. The fact is that the same underlying function that is used for Lua is used elsewhere and changing its behaviour the way you want would negatively affect other parts of the code. So decision has been made not to change it.

My I suggest a workaround for you. If you need or desire different formatting/display of the value, then do the formatting in the Lua code and use the lcd.drawText() function to output custom formatted value. That function should support all attributes.

closed 2015-10-18 by projectkk2glider

rdeanchurch, 2015-10-18:
projectkk2glider 
I don't 'love' the result, but don't hate it.
It just seemed a little inconsistent, but use throughout system makes it easy to understand the results of the report.
No sweat, thanks for the consideration.
]

PR #2977 (2015-10-17): Projectkk2glider/issue 2956 unit attributes
Closes #2956

Only a subset of attributes (INVERS | BOLD | BLINK | SMLSIZE) is passed along since the same function is also used on existing screens (especially DBLSIZE) and I did not want to mess that up.

If accepted, I will port it to 2.1 branch.
commit 7ad5c1e2 (2015-10-17): Fixes #2956: added possibility of INVERS, BOLD and BLINK attributes to Lua lcd.drawChannel() unit