opentx/opentx #2958: Input value with negativ sign (Input Invert)

ParkerEde, 2015-10-10:
Hi,
if you invert the input e.g. -100% aileron the trim works in the wrong direction.

I use Taranis X9D plus with 2.1.3 

regards
ParkerEde

kilrah, 2015-10-11:
That's intended.

closed 2015-10-11 by kilrah

tomkfly, 2015-10-11:
In most cases It is normal to Use negative in Mixes not in Inputs.

ParkerEde, 2015-10-12:
Hi kilrah and tomkfly. Thank you for your answers.
I create a delta mix for my jet. The elevator works in the correct direction the ailerons not. So I had to ways to correct die direction for the ailerons:
1. Enter the Mixers and change two values from positive to negative values
or
2. Enter the Inputs and change only one value from postive to negative

My decision was way 2, because I had to change only one value.
Sorry, I can't believe that it's intended that the trim works in another direction than the control sticks!
It would be very nice, if the trim function will provide the negative sign from inputs, so you can be sure the trim function is always right with the control sticks.

btw. the first flight with the Jet was not very nice..... ;-)
Kind regards
ParkerEde