opentx/opentx #2962: [OpenTX 2.1.3][Bug][TGY 9X] Switches continuously occupied

paulhood, 2015-10-11:
Dear all,
after flashing to OpenTX 2.1.3 I have problems with my switches. During boot sequense I get an error message as usally if the switches are not in correct position but they are.
Also within the menu they always have the same status (1) and couldn't used any more.
Wihin the simulator everything is fine. Hardware seams also to be okay, checked serveral times.
FW: opentx-9x-frsky-nofp-audio-gvars-potscroll-sp22-pgbar-de-2.1.3

projectkk2glider, 2015-10-11:
* Which was the previous OpenTX version that worked on your radio?
* Do I understand correctly, the switches (AIL, THR, etc,,) do not change status on main screen after moved.

paulhood, 2015-10-11:
- previous version Companion 1.52 with 2.1.1. 
- correct, but only two of them others are working fine

projectkk2glider, 2015-10-11:
Which two switches?

paulhood, 2015-10-11:
THR and AIL, both on the upper left side of the control unit

projectkk2glider, 2015-10-11:
Aren't those two switches affected by the FrSky telemetry mod? Are you sure you got your firmware options set properly?

paulhood, 2015-10-11:
I don't have any FrSky telemetry installed yet. Should I change the FW --> without telemetry and you think that should work?
As far as I know, I have set them correctly. How could I check the setup?

projectkk2glider, 2015-10-11:
Then remove `frsky` firmware option, download it again and reflash. Read http://openrcforums.com/forum/viewtopic.php?f=45&t=3450

closed 2015-10-11 by projectkk2glider