opentx/opentx #2966: LUA Script crashes on getting GPS position

tgreer, 2015-10-13:
Hi,
I have an issue with opentx where the LUA script crashes when it gets a GPS position.

@xdandys has patched this temporarily in his copy of opentx:
https://github.com/xdandys/opentx/commit/2289621e7ac8e7a93764ffc86d1f6c00c3869233

I was wondering if anyone could look at this and work out what needs to be done to fix it properly?

Thanks

Thomas

projectkk2glider, 2015-10-13:
Take a look at https://github.com/opentx/opentx/pull/2773#issuecomment-141445369 and other comment in that issue. You will se how to get latitude and longitude.

closed 2015-10-13 by projectkk2glider

hrenz, 2015-10-17:
GPS Koordinaten in LUA auslesen
"lat" und "lon" in Kleinbuchstben!

Beispiel:
gpsLatLon = getValue("GPS")
if (type(gpsLatLon) == "table") then
LocationLat = gpsLatLon["lat"]
LocationLon = gpsLatLon["lon"]
end