opentx/opentx #2969: Confirmation of Shutdown - Arduino Mega2560 and derivatives.

sloped-soarer, 2015-10-14:
HI.
Can we have a confirmation of shutdown on power off please for the Mega2560.
I believe the Taranis already has this.

Thanks.

projectkk2glider, 2015-10-17:
Is that even possible on mega2560? I mean does it have software controlled power switch?

bsongis, 2015-10-17:
Somebody asked me for Taranis (not X9E) as well. It should be easy to do, just a port but I am missing time to work on all boards in the same time.

projectkk2glider, 2015-10-17:
Bertrand, you concentrate on the new board, leave the small stuff to us...

bsongis, 2015-10-17:
The gruvin9x board has a soft power switch, not sure for the MEGA2560. @JivaroFAD knows (sorry for email I still didn't answer Franck, I have difficulties to do something else than Horus these days, so much work ...)

sloped-soarer, 2015-10-17:
Hi All.
Some people fit a software switch e.g.
http://jenkins.open-tx.org/MEGA2560/1_Listing-Schematics/
Some don't and then wonder why they have the jack plug icon !.
See
http://www.rcgroups.com/forums/showthread.php?t=2406337
Maybe a screen could be created for the latter to shut down gracefully.
It's your call.

projectkk2glider, 2015-10-17:
Ok, I will edit the title of this issue to include X9D also.

kilrah, 2015-10-17:
> Some don't and then wonder why they have the jack plug icon !.

> Maybe a screen could be created for the latter to shut down gracefully.

That is a whole different thing that would need another separate issue opened. The [!] appears only for people who haven't fitted a soft power switch, but a confirmation can only be done if there is a soft power switch. So a confirmation screen would not solve their issue.

So in addition there could be an option that disables soft power handling capability altogether, but you'd lose some functionality that way (persistent timers etc).

bsongis, 2015-10-17:
Yes an option is missing if some people have the soft power switch while others don't on MEGA2560

JivaroFAD, 2015-10-18:
Hi,
There's already a "Power management by soft-off circuitry" option in malkefile and Companion, since https://github.com/opentx/opentx/commit/61c7abf2146c1f23aa301ba2eb758935af3f1051, to avoid the "!" after a dirty power off.
But a popup menu to have a software clean shutdown would be nice for people who don't make the power circuitry, as it will allow to save timers.
@Bertrand : don't worry, I understand :)

sloped-soarer, 2015-10-19:
Hi @JivaroFAD.
The main reason for raising this issue was that if i just nudged the power switch, the tx might shutdown.
IDK if the code does a persistance check or whether the hardware should filter for such situations.
I'd rather have the option to confirm or cancel.

sloped-soarer, 2016-02-26:
Hi @JivaroFAD.
Maybe implement something like this ...
It works for me, but my hardware is a little different to yours, but i think the logic is still the same.

Lines 122 to 145.

https://github.com/sloped-soarer/OpenTX-MPX-Royal-Mega2560/blob/master/opentx/src/targets/mega2560/evo_mega2560.cpp

JivaroFAD, 2016-02-26:
Hi,
Nice, thanks ;) I have to add voice option to M2560 project (https://github.com/Ingwie/opentx, see also : https://youtu.be/GIsYYxqEM9g), I will probably add your work in the same time.

sloped-soarer, 2016-02-26:
@JivaroFAD.
I love the voice option ... i can feel a new pcb comming although i'm trying to resist any more electronic projects.

Just added the schematic pdf's.
https://github.com/sloped-soarer/OpenTX-MPX-Royal-Mega2560/blob/master/opentx/src/targets/mega2560/evo_mega2560.cpp

Best wishes.

closed 2020-07-07 by stale[bot]