opentx/opentx #2971: Access violation when launching Simulator from Mixes tab

cass3825, 2015-10-15:
[image]

projectkk2glider, 2015-10-17:
Which version of Companion? Anything particular about the model that causes the crash or does it crash if you just create and simulate empty model? If only particular model, then post it here.

cass3825, 2015-10-17:
Version 2.1.3
Simulator crashes when using logic switch to enable a Flight Mode.  Confirmed with blank model using L1 switch entry on Flight Mode 1.

projectkk2glider, 2015-10-17:
Damn, cant reproduce on Linux. Perhaps the change to strip companion binary (`SET(CPACK_STRIP_FILES TRUE )`) in 51b8d70cc650605f85a42e37a74b07b5ccddd26f was not so smart (it was made by me :stuck_out_tongue: ). Now the backtrace is empty and not very useful.

projectkk2glider, 2015-10-17:
@cass3825 which radio are you simulating?

bsongis, 2015-10-17:
Yes strip files was a good idea :) I will try to find another minute for this one.

cass3825, 2015-10-17:
@projectkk2glider taranis plus.

mpaperno, 2017-04-11:
Cannot recreate in `next`, most likely fixed by the numerous other changes in 2.2.

closed 2017-05-11 by mpaperno