opentx/opentx #2972: [OpenTX 2.0.x & 2.1.x][Bug ??] Slow option in mixer doesn't behave as expected

LapinFou, 2015-10-15:
Hi folks,

Last night I tried to help somebody on RCGroups.
Explications + EEPE available here:
http://www.rcgroups.com/forums/showthread.php?t=2521970

IMPORTANT: I ran the tests only under Companion (2.0.17 & 2.1.3). I didn't test on the radio itself.
I caught something which doesn't look right. In the example below (with a slow parameter of 4s), the CH1 will travel from -100% to +100% in 2s, instead the expected 4s ?!? Looks like there is a ratio of 2 somewhere, but I don't understand why... :flushed: 
[image]

tomkfly, 2015-10-15:
You have the delay on a mix with a weight of 200, so it is going to take 4 seconds to travel from -200 to +200,

kilrah, 2015-10-15:
Once again, slow works on the change of the source value (of which there is none with MAX), and shouldn't be used if a switch is set.

LapinFou, 2015-10-15:
@tomkfly You statement is not correct. If your replace the +200% weight with a +100% one, you will get the same result.
@kilrah I wasn't aware of this restriction.

tomkfly, 2015-10-15:
If set to 100 it will only travel half way and will take two seconds to travel, or at least it does on my version of Companion (2.0.17).

LapinFou, 2015-10-15:
It will travel half way (from -100% to 0%) in 2s with a slow parameter at 4s!
Just tested in Companion 2.0.17
I added a logical switch:
L1: OR SH↓ SH↓ Delay 2s
L1 goes true exactly at the same time it reaches 0%.

kilrah, 2015-10-15:
> It will travel half way (from -100% to 0%) in 2s with a slow parameter at 4s!

That is correct then, as the slow parameter is "how long should it take to travel from -100 to +100".

LapinFou, 2015-10-15:
Thanks to all of you for you explanation. :smiley: 
Last question @kilrah : when you wrote "Once again slow works on the change of the source value (of which there is none with MAX), and shouldn't be used if a switch is set." Do you mean there is a better way to do it, or do you mean it is dangerous to do it this way?

projectkk2glider, 2015-10-17:
@LapinFou The slow should be used on source change and not on a mixer line that is enabled by the switch.

So if you have a switch controlled mixer line, then create a slow on another mixer channel that has for the source the original mixer channel. That way the source will change and slow will work as expected.

I think this issue can be closed.

closed 2015-10-17 by projectkk2glider

LapinFou, 2015-10-17:
@projectkk2glider 
Thanks for your explanations. Indeed in the way I was using it (with MAX), I could observe some weird behavior.