opentx/opentx #2973: Warning when 2 modules are running in the same time

bsongis, 2015-10-15:
Everything in the title: a warning if you activate a module but there is already another one activated. Good or bad idea?

hrenz, 2015-10-15:
I  have  the internal  XJT and an extern XJT running same time 
do you mean  this  combination?
or 2 different Moduls connnected at external Modul socket

bsongis, 2015-10-16:
Yes I meant this combination. But external != XJT

tbs-trappy, 2015-10-16:
the issue is that sometimes beginners do not realize that their internal and external (mostly UHF) modules are running. The most popular frequency for UHF pilots is 2.4GHz video, which is totally destroyed by the close proximity between the internal TX module and the video receiver. The idea is to make sure that the internal module is not left running unintentionally when a external module is turned on.

bsongis, 2015-10-16:
We could add a warning at the time the user chooses an external module (except XJT).

tbs-trappy, 2015-10-16:
yes, this would satisfy most issues without taking away any flexibility

griebd, 2015-10-16:
+1, in my case I use a hacked DSM module (for some cheap helis and planes) and been warned would at least save some tx bat power (not that I have forgot any so far, I think :stuck_out_tongue_winking_eye:), if nothing else.

janekx, 2015-10-17:
I suggest to do it but have posibility this warning permanently disable on model basis but have it by default ON.

kilrah, 2015-10-17:
Again it would show up only at the moment you enable one module AND another is already on. You should not see it more than once when you set your model up.

janekx, 2015-10-17:
Ok one time

closed 2020-07-07 by stale[bot]