opentx/opentx #2975: mapping of transmitted channels and customized failsafe positions is wrong

Nihonski, 2015-10-16:
Hello, please have a look at following problem.
Environement: FrSky X9D plus B02, X8R Receiver, set up by OpenTX Companion 2.0.17
I can set the Companion, first page of any model, to start transmitting from channel 10 to channel 18, for instance. This is a normal function and this works correctly. Channel 10 comes out at the receiver position 1, channel 11 comes out at receiver position 2 and so on.
So I would expect that the failsafe position 1 comes out at the receiver position 1 and failsafe position 2 comes at receiver position 2 and so on. (There is no meaning to set a failsafe position for a channel which is not transmitted). But this is wrong. 
Actually I have to set the failsafe position 10 to come out at Receiver position 1, a.s.o..
And because of that, I cannot set custome failsafe for channel 17, 18, which is reveiver output 7 and 8, as it does not exist in Companion.
I suggest to map the failsafe positions and transitted channels correctly, or expand the failsafe positions in the Companion to 32 positions.
I believe this causes un-knowingly wrong failsafe settings and possible crashes.
with best regards, Nihonski

projectkk2glider, 2015-10-17:
>I suggest to map the failsafe positions and transitted channels correctly, or expand the failsafe positions in the Companion to 32 positions.

Expansion to 32 channels would take up unnecessary space and complicate things. The correct solution is to map the fail-safe position 1 to transmitted channel 1. Anybody has a problem with that?

kilrah, 2015-10-17:
That is what it should be, if it isn't then it's a bug.

projectkk2glider, 2015-10-17:
I have just verified this. I think the current state is correct and should not be changed.

You can set custom failsafe for all 32 channels, so if you start sending CH10 then set failsafe for CH10. You should know that channel 10 is coming out of CH1 on receiver because you have set it so. There is no point in hiding this from the user. 

The change would only make sense if there were no option to set the failsafe position for all channels. That was my initial thought, since I have never used this. But as noted above all is already in place and working. I vote for closure of this issue.

Nihonski, 2015-10-18:
I can set custom failsafe for 32 channels on the Taranis X9D+ itself, but not in the Companion on my PC. There are only 16. Invonvenient restriction.
The custom failsafe setting on the Taranis works for all 32 channels, verified by actuall test, but it is not indicated for the channels 17 to 32, these settings alsways show 0 after closing the tab, no matter what you keyed in before. But the failsafe values for all 32 channels stored in the receivers (after waitig around 9s) are correct. Thats a small, but irritating bug.

projectkk2glider, 2015-10-18:
Confirmed on 2.1.3, there are two problems:
* in Companion: only first 16 channels can be set when the custom failsafe is enabled
* on the radio: only CH1-8 can be set, CH9-CH32 can not be set, long pres on ENT key does nothing.

kilrah, 2015-10-18:
> I have just verified this. I think the current state is correct and should not be changed.
You can set custom failsafe for all 32 channels, so if you start sending CH10 then set failsafe for CH10. 

That is no good, as the failsafe settings are stored per module. So they must match the module outputs, not the radio's logical channel.

projectkk2glider, 2015-10-18:
But Kilrah consider this situation:
* radio sends channels 10-25 to the receiver (starting channel is 10) and they come out on receiver at positions 1-16
* you suggest that CH1 failsafe value should be sent for the first channel.
* But when you go to custom falilsafe screen (on the radio) and set the values using sticks the CH1 value is taken from the first mixer, not the CH10 mixer. But the current CH10 mixer output value should be used to set the failsafe position for that channel.

Nihonski, 2015-10-29:
I think first of all the bug on the radio should be fixed: 
"only CH1-8 can be set, CH9-CH32 can not be set, long pres on ENT key does nothing"

Second; it would be best to exand the Companion Custom Failsafe Setting to 32 channels and all of them should be transmitted (maybe takes longer than now) and memorized in the receivers. 
I am not sure whether that would need FW updates of FrSky RF modules, but this would match best to the OpenTX philosophie that everything is freely programmable. Users who understand Companion will also understand how to map the mixer channels correctly with failsafe positions.
I think this would be also the safest way for upgrading OpenTX on radios, without currupting existing radio user settings.

As for me, I will use receiver failsafe for the time-being, although it is considered old fashioned, but it works flaw-less.

kilrah, 2015-10-30:
> "only CH1-8 can be set, CH9-CH32 can not be set, long pres on ENT key does nothing"

Actually CH9-16 can be set if the channel range for the given module includes them, this is correct.
The bug is that the 2nd page (17-32) should not be accessible (the bug comes from the fact that this is handled by the same menu as the channel monitor, and the case was not separated).

> radio sends channels 10-25 to the receiver (starting channel is 10) and they come out on receiver at positions 1-16
you suggest that CH1 failsafe value should be sent for the first channel.
But when you go to custom falilsafe screen (on the radio) and set the values using sticks the CH1 value is taken from the first mixer, not the CH10 mixer. 

Failsafe settings are module, and thus receiver specific. So setting "CH1" failsafe on say the internal XJT should affect CH1 output of the receiver bound to it, even if it's the radio's logical CH10. So yes a shift is missing, the radio channel used to set a failsafe value should be shifted by the number in channel start value.

Nihonski, 2015-10-30:
I agree with that, the second page (17-32) does not need to be seen on the radio, if the mapping/shifting is done correctly.
One more thing to mention:
Presently I can set on radio to Start transmitting with channel 32 (for example) and transmit 16 channels. That means the logical channels 32-48 would be transmitted, which do not exist.
Guess the Start setting needs to be limited to 16.

kilrah, 2015-10-30:
> Presently I can set on radio to Start transmitting with channel 32 (for example) and transmit 16 channels. 

Is that on companion? I just checked on the radio and the range settings are constrained correclty.

Nihonski, 2015-10-30:
Right, that is on the Companion. Sorry for wrong explanation.

closed 2015-10-30 by kilrah

kilrah, 2015-10-30:
OK, should be fixed both on 2.0 and 2.1. Tested on both versions, only on internal module asI don't have an external XJT at hand.
I've updated the radio UI to only show channels that are transmitted for clarity. Should now be quite obvious that the mapping matches the receiver labelling.

Nihonski, 2015-10-30:
Thank you very much, Looks good.
If I may ask, which Version of 2.0.x and 2.1.x will it be implemented?

kilrah, 2015-10-30:
Well the next releases of both branches.

Nihonski, 2015-10-30:
So the 2.0.x is still supported.
Danke nochmals und viele Grüße in die Schweiz aus Japan.

commit 08ff104d (2015-10-30): Correct failsafe mapping (fix #2975)
commit f2d316b8 (2015-10-30): Correct failsafe mapping (fix #2975)