opentx/opentx #2980: [Taranis B][Special Functions] "PlayTrack","Track selection not working" popup menu

483917, 2015-10-18:
Sorry for my bad english, hope you understand my explanation ?

Windows 10 pro swedish
Companion 2.1.4N494
Taranis B with , "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N494.bin"

On Taranis B.
I cant change Track ( "Play Track" function in "Special Functions" ) on any of my models ?
The popup menu dont work, and if i select the "Lua Script" function its the same, no popup menu
in Script selection.
If i then exit out and try to change model ,i get no popup menu in model selction list.

I have to restart the radio to select an other model 

In Companion Simulator all is working normal

My EEPROM:
https://www.dropbox.com/s/dh1pqasro5o14ox/eeprom-2015-10-18-174447.bin?dl=0

projectkk2glider, 2015-10-18:
Wow, did it work in any of the previous versions or to ask differently, do you have any idea which version broke this?

483917, 2015-10-18:
I just tested "opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N468.bin"
and same behavior hmm

projectkk2glider, 2015-10-18:
I can't reproduce. But you description suggest you have a problem with the SD card. Do the sounds work, for example if you move trim to the centre, do you hear the "trim center"?

483917, 2015-10-18:
Yes my sounds are working , tested trim center and others to seems ok

Strange ?

I will do some more tests later
when I get home

Skickat från min mobil

:-)

483917, 2015-10-18:
@projectkk2glider 
I have done a load of tests now and wonder , is there a limit on how many files you can have
in folder, \SOUNDS\en\*.wav  ?
I even changed to a new SD-card, but did not work
then i deleted all my own made soundfiles and the popup list was ok again

I have 260 sound files in \SOUNDS\en\ , and mabye thats why the popup list dont work ??

bsongis, 2015-10-19:
This is a nice guess, would you try with 100 files max? If it works, then there is a 8bits cast somewhere.

projectkk2glider, 2015-10-19:
I have ~170 sound files.

bsongis, 2015-10-19:
Perhaps a limit around 256? Or one file with special chars?

projectkk2glider, 2015-10-19:
No problem with 280 files in Companion radio simulator.

483917, 2015-10-19:
Ok i have done more testing and with 258 files all is OK
but if i add one more file = 259 then the popup menu do not work ?

bsongis, 2015-10-19:
I think we are close to understand, thanks for this information!

bsongis, 2015-10-25:
Not reproducible on simu it seems

483917, 2015-10-25:
@bsongis 
To day i did som more testing on Taranis
Tested with firmware:
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N494.bin"
and
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N502.bin"

And i also changed my original Taranis SD-card to a compleatly new 2gig SD-card

Then i started testing, and i cant reproduce the "PlayTrack" popup list crash ?? 
Mabye my original Taranis SD-card had some defect ?
All files are the same on old and new SD-card

so i think its OK to close this issue, the popup list is working normal

bsongis, 2015-10-25:
Which version had the bug?

483917, 2015-10-25:
This version had the bug.
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N494.bin"
with my original Taranis SD-card and more then 258 files in folder, \SOUNDS\en\

But after i changed my SD-card to a new one ,the bug is gone ?? popup list work normal
i have tested
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N494.bin"
and
"opentx-taranis-haptic-timer3-lua-mixersmon-sqt5font-en-2.1.4N502.bin

i also noticed that the conection time, Taranis-USB to PC "win10 and WinXP" was mutch faster
with my new SD-card

bsongis, 2015-10-25:
Ok, let's say it is the SD then!

closed 2015-10-25 by bsongis

projectkk2glider, 2015-10-25:
@483917 run that SD card trough a disk checking tool, perhaps the file system is corrupt.

483917, 2015-10-25:
yes i will test the SD card later to night ,and perhaps find som error