opentx/opentx #2982: Preflight check switch individual position 2.1.3, 2.0.17

janekx, 2015-10-19:
I suggest to able to set individual switch position in preflight check for every model, now there is only arrow up and off (no arrow). 
There should be cycle of arrow up, middle, arrow down and off (no arrow).
Pots have position check so why not switches?

projectkk2glider, 2015-10-19:
There is a simpler way, just set switches to their desired position and long press on `<` symbol. All switches will remember their current position.

janekx, 2015-10-19:
I see it works, thanks, but is it hard to find this function when someone will expect changing switch each alone....

projectkk2glider, 2015-10-19:
It has been like that for a while (I think from 2.0).

lepicfoto, 2015-10-25:
sorry to ask, i can not find "long press on < symbol." on my taranis, how can i make my taranis remember the switch pos

Best regards
Jesper

janekx, 2015-10-25:
It mean Enter long press to set

projectkk2glider, 2015-10-25:
> i can not find "long press on < symbol." on my taranis

True, that graphic was there in previous versions (at least I remember it like that). Now just as Janek wrote, just long press ENT when the whole row is selected.

kilrah, 2015-10-25:
> It has been like that for a while (I think from 2.0).

It's been like that since OpenTX exists actually :)

closed 2015-10-25 by kilrah