opentx/opentx #2984: Negative timer value not possible in Logical Switches

rbak69, 2015-10-20:
I use a logical switch in order to trigger special function to start playing the timer value every 10 seconds when the value is smaller then -00:09. This will result in starting to play the announcements when it is -10 seconds and from then on repeats that every 10 seconds. I use Timer 2 for this and configured the timer to start counting down from 10 minutes. During the count down from 10 minutes I use the normal Minute call functionality to play the timer value every minute.

In order to achieve the additional (negative) announcements  I have done the following in the openTX firmware for my Taranis:

Logical Switches
L1    a<x    Tmr2    -00:09

Special Functions
SF1    L1    PlayValue    Tmr2    10s

So as soon as the timer value goes below 9 seconds L1 is true and the next announcement is played at -10 seconds and then continuing every 10 seconds (-10 seconds, -20 seconds, -30 seconds, etc.). This works very well and I am very happy with this.

The only annoying thing is that Companion does not support entering a negative value for the timer value in Companion. It reads the eepe file just right and even the simulator works fine with the eepe file retrieved from my Taranis. But the value in the Logical Switch shows as 00:00:00 [h:m:s]. If I would like to set this up with Companion itself it will not work as it does not allow to change the value to a negative value.

This has been in there for multiple 2.0 versions up to 2.0.17 and also still is the case for version 2.1.13. For all the corresponding firmware versions for the Taranis itself it works fine.

Would like to see this bug fixed in Companion as well.

kilrah, 2015-11-07:
Companion actually allows 0 to 60min, while radio allows +/-30min.
30min doesn't seem like much...

rbak69, 2016-01-25:
Is there any idea when this bug will be solved? The milestone has changed a few times since I have raised the issue and no one has been assigned to resolve the issue yet as it seems. Just curious when I can expect this will be fixed in companion.

projectkk2glider, 2016-01-25:
The QT control that we use does not support negative time values! Bummer.

rbak69, 2016-02-02:
So is there a way around this? I understand that the QTimeEdit control does not support negative values (which makes sense) but is there no other control that can be used for this? After all we are not setting the actual time (the clock).

Sofar the functionality on OptenTx works correctly so the issue is mainly annoying as Companion does not allow you to set this correctly for the way I use it. Would be nice if an alternative control could be used to solve this issue so we can also configure this correctly in Companion.

PR #5437 (2017-12-19): [Companion] Adjust source value ranges and editors in LS/CF/Telemetry
Went down a rabbit hole with regards to timer and RTC ranges and how they're used/limited/edited in regards to LS (comparison functions), CF (set timerN action), and telemetry display bar limits.  Seems there's a bit of confusion here both in Companion and on radio (probably due to legacy issues).

Fixes #2984
Fixes #5362  (also negative values were not possible)
Fixes #5374
Fixes #5375
Fixes #5527
Fixes timer and RTC time ranges to comply with firmware limits.
Fixes not being able to use RTC time in LSs.
Fixes newly-added L [... cut]
closed 2017-12-19 by bsongis