opentx/opentx #2985: Update Lua Reference Guide for OpenTx2.1

r41065, 2015-10-20:
changing my Taranis from OpenTx 2.0.17 to 2.1.x found some lines in my Telemetry script which are not longer working correct.
for example:
  local modelinfo = model.getInfo()
  lcd.drawText(10,20,modelinfo.name,MIDSIZE)    --> works correct
  lcd.drawNumber(10,30,modelinfo.id,MIDSIZE)    --> returns Syntax Error

local fm = getValue("flight-mode")  --> returns always "0"

lcd.drawChannel(80, 20,"clock", MIDSIZE) --> returns Minutes from midnight, not "hours:minutes" as in 2.0.17

I wonder that they are not working for the new Version because I copied these Functions from the Reference Guide for OpenTx 2.0 .
I can't find any Documentation to check if there is a new syntax or if this are open points for Lua scripting in OpenTx 2.1

Ralf

projectkk2glider, 2015-10-22:
https://github.com/dsbeach/lua-reference-guide is I thing the most recent version of documentation.

r41065, 2015-10-22:
Sorry, but i get only:

Come back later!
This book is still in preparation by its author.

projectkk2glider, 2015-10-22:
I see, here is a link to the fork https://www.gitbook.com/book/dsbeach/opentx-lua-reference-guide/details by @dsbeach which was not yet merged into the main OpenTX repository.

dsbeach, 2015-10-22:
I'm here and still working on it.  Should I create a pull request for the current state?

I've been busy putting together a debugging environment for the companion so I can hopefully correct whatever defects are uncovered by bringing the documentation up to date.

r41065, 2015-10-22:
Thank you for Updated Link, now i can do my Test
and yes, there are some new functions for the scripts..

closed 2015-10-22 by r41065

bsongis, 2015-10-22:
@dsbeach Yes for the pull request, it would be nice, I hope it will work, it will be the first pull request on the doc

dsbeach, 2015-10-22:
I'm using the gitbook editor which generates a ton of commits.  I'm new to git and wonder if there is a easy way to consolidate all the commits down to one per file.

Also, should I plaster a 'DRAFT' watermark on the cover page, or are you OK as it is?

bsongis, 2015-10-23:
I think it is already OK!