opentx/opentx #2986: Is it possible to get information about flight mode ((ALTCHOLD, stability, RTL, etc.) on the screen Turnigy 9XR PRO?

darektv, 2015-10-21:
Is it possible to get information about flight mode ((ALTCHOLD, stability, RTL, etc.) on the screen Turnigy 9XR PRO?

I installed version opentx-9xrpro v 217
connection (Ardupilot -> Arduino Pro Mini -> FrSky receiver -> er9x-based transmitter) (https://code.google.com/p/er9x-frsky-mavlink/)
I have all the data telemertyczne (GPS, ALT, HDG etc.) as digits
everything works OK
necessary information about flight mode
how to do it ??
It would be very in handy
You would not need to remember the switches

I searched new sensors on the radio
when changing flight mode on the radio, changing only the parameters of sensors 000F and 0001D

darek

kilrah, 2015-10-22:
Have a look on forums, many use the radio's flight mode feature to display the name of the mode the current switch combination results in. 
Native MAVLINK telemetry is not supported on platforms other than the 9X at this point (#2354).

closed 2015-10-22 by kilrah

darektv, 2015-10-22:
Hello
Thank you for your response
if there are opportunities for the development platform opentx in that direction?
Darek

bsongis, 2015-10-22:
Yes the flight mode will soon have his own S.PORT ID for that purpose. Just a matter of time, we are too busy, a lot of changes in OpenTX these days

darektv, 2015-10-23:
Hello
Thank you for your response
I'm looking forward
I will help in testing
I have set: Turnigy 9XR PRO Quanum NOVA multirotor controler (compatible with Ardupilot) Arduino pro mini soft to convert mavlink to FrSky (project https://code.google.com/p/er9x-frsky-mavlink/) FrSky receiver D8R-II plus
darek

projectkk2glider, 2015-10-23:
@darektv you might be interested in my fork of the project available at https://github.com/projectkk2glider/APM-Mavlink-to-FrSky. It fixes several problems I had with the original project.