opentx/opentx #2987: companion/opentx 2.1.3 - Taranis not recognized when plugged in on linux PC

dk7xe, 2015-10-23:
since moving from 2.0.17 to the 2.1.x version i had always problems that the Taranis is not always recognized by openTX resulting that the SDcard sync is working but not the read/write parameters.

I did an investigation on the udev rule (https://github.com/opentx/opentx/blob/master/companion/targets/linux/45-companion-taranis.rules) and found that the deviceID is not corresponding to the one the Taranis actually has.

45-companion-taranis.rules has as content
*SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE:="0666";*

Wile *lsusb* on my Linux system is showing that the Taranis is having ProductID "5720" 

*Bus 003 Device 002: ID 0483:5720 STMicroelectronics*

Changing the 45-companion-taranis.rules content to 

*SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5720", MODE:="0666";*

solved the problem for me. Taranis is now well recognized by the companion software with every plug in to my Linux PC.

To structurally solve the issue i suggest to add all the possible ProductID's to the udev rule with seperate entries e.g. that 45-companion-taranis.rules contains both lines

*SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE:="0666";*
*SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5720", MODE:="0666";*

or there is a way to force the Taranis via opentx to report always the "df11" as productID to the USB system.

**My configuration**
Taranis plus with openTX 2.1.3
Asus EeePC 1201N with openSuse 13.2 64bit 
companion 2.1.3

kilrah, 2015-10-23:
0483:df11 refers to the processor's internal bootloader that is used for flashing using dfu-util when power is off.

When power is on in bootloader mode the IDs are obviously different, but no udev rules should be needed to access the simple mass storage drive they relate to.

dk7xe, 2015-10-23:
Hi Andre, hmm then it seems my solution is maybe not targeting the real rootcause.

May you please elaborate a bit more how companion is detecting the presence of the opentx device?

Strange is that it works perfectly on my Windows tablet but under openSuse i have issues. Maybe i need help by the package maintainer of the suse companion package if it is specific to that. But for this it would help to understand how presence of the Taranis is detected.

Thank you!
Gerald

projectkk2glider, 2015-10-23:
>May you please elaborate a bit more how companion is detecting the presence of the opentx device?

On linux: loop over all mounted file-systems and return the first path that contains EEPROM.BIN in the root directory.

dk7xe, 2015-10-23:
Thanx that helps already. Is there a limit on path depth or a another search restriction e.g. to look only for mounted devices in /mnt ?

Gerald

projectkk2glider, 2015-10-23:
There should be no limit, the process is two step:
* first a list of mounted filesystems is retrieved in read_file_system_list() https://github.com/opentx/opentx/blob/7a01f04f498df51771ab3ffd35ea6acb6a7b3217/companion/src/mountlist.cpp#L377
* then that list is used when searching for existence of EEPROM.BIN (if you try to read/write model setttins) or FIRMWARE.BIN if you try to flash the radio. Search function is findMassstoragePath() https://github.com/opentx/opentx/blob/7a01f04f498df51771ab3ffd35ea6acb6a7b3217/companion/src/radiointerface.cpp#L380

Btw, looks like we have a memory leak, the `entry` linked list (in  findMassstoragePath() ) must be free-ed before the function returns.

dk7xe, 2015-10-24:
I did now several test with the udev rule modified and without.
With idProduct = 5720
/dev/sdd (TARANIS) got immediately mounted when plugged before /dev/sdc (SDcard).

When i changed the idProduct back to df11 the strange behaviour came back.

What i see when running *mount* several times after plugging in the taranis is that 
/dev/sdc (SDcard) is mounted immedeately but
/dev/sdd (TARANIS) not.
Its taking between 30-120sec until sdd gets mounted.
That explains why the SDcard sync is working while data retrieve not .

But i don't have any idea why it is like this.

Gerald

dk7xe, 2015-10-24:
Thank you guys for the explanations.
I think the delay in mounting the TARANIS is specific to my system by any reason. Therefore i will follow up on openSuse side.
As a workaround i will add an udev rule with idProduct="5720" for the time being.

But i guess the memory leak should be looked at.

Thank you!
Gerald

bsongis, 2015-10-24:
You may check with wireshark what is going on the USB port

projectkk2glider, 2015-10-24:
There is also the issue #1152 you might look at (warning: a lot of text).

projectkk2glider, 2015-10-24:
@dk7xe You can look at the syslog when you connect the Taranis. It might show a problem. Mine is here. From the connection to the mount takes only 4 seconds (2GB card). Also check if filesystem on SD card is corrupt and needs fsck.

```
$ tail -n0 -f /var/log/syslog
Oct 24 20:51:23 mambo kernel: [180405.189630] usb 2-1.8: new full-speed USB device number 71 using ehci-pci
Oct 24 20:51:23 mambo kernel: [180405.284079] usb 2-1.8: New USB device found, idVendor=0483, idProduct=5720
Oct 24 20:51:23 mambo kernel: [180405.284084] usb 2-1.8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 24 20:51:23 mambo kernel: [180405.284087] usb 2-1.8: Product: FrSky Taranis Bootloader
Oct 24 20:51:23 mambo kernel: [180405.284089] usb 2-1.8: Manufacturer: FrSky
Oct 24 20:51:23 mambo kernel: [180405.284091] usb 2-1.8: SerialNumber: 00000000001B
Oct 24 20:51:23 mambo kernel: [180405.284709] usb-storage 2-1.8:1.0: USB Mass Storage device detected
Oct 24 20:51:23 mambo kernel: [180405.284884] scsi17 : usb-storage 2-1.8:1.0
Oct 24 20:51:23 mambo mtp-probe: checking bus 2, device 71: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8"
Oct 24 20:51:23 mambo mtp-probe: bus: 2, device: 71 was not an MTP device
Oct 24 20:51:24 mambo kernel: [180406.282088] scsi 17:0:0:0: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
Oct 24 20:51:24 mambo kernel: [180406.282658] scsi 17:0:0:1: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
Oct 24 20:51:24 mambo kernel: [180406.282975] sd 17:0:0:0: Attached scsi generic sg3 type 0
Oct 24 20:51:24 mambo kernel: [180406.283242] sd 17:0:0:1: Attached scsi generic sg4 type 0
Oct 24 20:51:24 mambo kernel: [180406.283918] sd 17:0:0:0: [sdd] 3842048 512-byte logical blocks: (1.96 GB/1.83 GiB)
Oct 24 20:51:24 mambo kernel: [180406.289315] sd 17:0:0:1: [sde] 1091 512-byte logical blocks: (558 kB/545 KiB)
Oct 24 20:51:24 mambo kernel: [180406.289595] sd 17:0:0:0: [sdd] Write Protect is off
Oct 24 20:51:24 mambo kernel: [180406.289599] sd 17:0:0:0: [sdd] Mode Sense: 00 00 00 00
Oct 24 20:51:24 mambo kernel: [180406.290304] sd 17:0:0:1: [sde] Write Protect is off
Oct 24 20:51:24 mambo kernel: [180406.290310] sd 17:0:0:1: [sde] Mode Sense: 00 00 00 00
Oct 24 20:51:24 mambo kernel: [180406.290954] sd 17:0:0:0: [sdd] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.290958] sd 17:0:0:0: [sdd] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.291584] sd 17:0:0:1: [sde] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.291589] sd 17:0:0:1: [sde] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.295950] sd 17:0:0:0: [sdd] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.295953] sd 17:0:0:0: [sdd] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.296576] sd 17:0:0:1: [sde] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.296578] sd 17:0:0:1: [sde] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.310613]  sdd: sdd1
Oct 24 20:51:24 mambo kernel: [180406.376730]  sde:
Oct 24 20:51:24 mambo kernel: [180406.378185] sd 17:0:0:0: [sdd] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.378187] sd 17:0:0:0: [sdd] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.378189] sd 17:0:0:0: [sdd] Attached SCSI removable disk
Oct 24 20:51:24 mambo kernel: [180406.444779] sd 17:0:0:1: [sde] Asking for cache data failed
Oct 24 20:51:24 mambo kernel: [180406.444785] sd 17:0:0:1: [sde] Assuming drive cache: write through
Oct 24 20:51:24 mambo kernel: [180406.444788] sd 17:0:0:1: [sde] Attached SCSI removable disk
Oct 24 20:51:25 mambo udisksd[2158]: Mounted /dev/sde at /media/damjan/TARANIS on behalf of uid 1000
Oct 24 20:51:27 mambo udisksd[2158]: Mounted /dev/sdd1 at /media/damjan/4433-2211 on behalf of uid 1000
```

dk7xe, 2015-10-27:
I just gave it another try but then it was working immediately.. 

log did not show anything beside sdc1 not properly unmounted. But anyhow that was when it worked.
I will come back after some further tests.

Hmm wireshark.. is installed but i have never used it so far.  I'll give it a try.

015-10-27T06:51:10.407829+01:00 gppc kernel: [  220.015116] usb 4-1: new full-speed USB device number 2 using ohci-pci
2015-10-27T06:51:10.610813+01:00 gppc kernel: [  220.218201] usb 4-1: New USB device found, idVendor=0483, idProduct=5720
2015-10-27T06:51:10.610878+01:00 gppc kernel: [  220.218216] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2015-10-27T06:51:10.610890+01:00 gppc kernel: [  220.218226] usb 4-1: Product: FrSky Taranis Bootloader
2015-10-27T06:51:10.610898+01:00 gppc kernel: [  220.218236] usb 4-1: Manufacturer: FrSky
2015-10-27T06:51:10.610906+01:00 gppc kernel: [  220.218244] usb 4-1: SerialNumber: 00000000001B
2015-10-27T06:51:10.619824+01:00 gppc kernel: [  220.227396] usb-storage 4-1:1.0: USB Mass Storage device detected
2015-10-27T06:51:10.620835+01:00 gppc kernel: [  220.228397] scsi7 : usb-storage 4-1:1.0
2015-10-27T06:51:11.631805+01:00 gppc kernel: [  221.239269] scsi 7:0:0:0: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-10-27T06:51:11.632780+01:00 gppc kernel: [  221.240226] sd 7:0:0:0: Attached scsi generic sg2 type 0
2015-10-27T06:51:11.640114+01:00 gppc kernel: [  221.247398] scsi 7:0:0:1: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-10-27T06:51:11.640814+01:00 gppc kernel: [  221.248929] sd 7:0:0:1: Attached scsi generic sg3 type 0
2015-10-27T06:51:11.650818+01:00 gppc kernel: [  221.258320] sd 7:0:0:0: [sdc] 7862272 512-byte logical blocks: (4.02 GB/3.74 GiB)
2015-10-27T06:51:11.661805+01:00 gppc kernel: [  221.269285] sd 7:0:0:0: [sdc] Write Protect is off
2015-10-27T06:51:11.661856+01:00 gppc kernel: [  221.269302] sd 7:0:0:0: [sdc] Mode Sense: 00 00 00 00
2015-10-27T06:51:11.668760+01:00 gppc kernel: [  221.276264] sd 7:0:0:1: [sdd] 1091 512-byte logical blocks: (558 kB/545 KiB)
2015-10-27T06:51:11.675809+01:00 gppc kernel: [  221.283281] sd 7:0:0:0: [sdc] Asking for cache data failed
2015-10-27T06:51:11.675873+01:00 gppc kernel: [  221.283296] sd 7:0:0:0: [sdc] Assuming drive cache: write through
2015-10-27T06:51:11.682780+01:00 gppc kernel: [  221.290266] sd 7:0:0:1: [sdd] Write Protect is off
2015-10-27T06:51:11.682843+01:00 gppc kernel: [  221.290279] sd 7:0:0:1: [sdd] Mode Sense: 00 00 00 00
2015-10-27T06:51:11.695026+01:00 gppc kernel: [  221.301262] sd 7:0:0:1: [sdd] Asking for cache data failed
2015-10-27T06:51:11.695086+01:00 gppc kernel: [  221.301273] sd 7:0:0:1: [sdd] Assuming drive cache: write through
2015-10-27T06:51:11.766787+01:00 gppc kernel: [  221.374353]  sdc: sdc1
2015-10-27T06:51:11.913037+01:00 gppc kernel: [  221.520281]  sdd:
2015-10-27T06:51:11.919766+01:00 gppc kernel: [  221.527228] sd 7:0:0:0: [sdc] Attached SCSI removable disk
2015-10-27T06:51:12.005751+01:00 gppc kernel: [  221.613245] sd 7:0:0:1: [sdd] Attached SCSI removable disk
2015-10-27T06:51:13.617558+01:00 gppc udisksd[1510]: Mounted /dev/sdd at /run/media/gpeklar/TARANIS on behalf of uid 1000
2015-10-27T06:51:13.628946+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1296]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/index.bdmv
2015-10-27T06:51:13.631114+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1296]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/BACKUP/index.bdmv
2015-10-27T06:51:16.260731+01:00 gppc kernel: [  225.863205] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
2015-10-27T06:51:16.261429+01:00 gppc udisksd[1510]: Mounted /dev/sdc1 at /run/media/gpeklar/OTX_SD on behalf of uid 1000
2015-10-27T06:51:16.277111+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1296]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/OTX_SD/BDMV/index.bdmv
2015-10-27T06:51:16.280442+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1296]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/OTX_SD/BDMV/BACKUP/index.bdmv

projectkk2glider, 2015-10-27:
Looks like `UDisks2VolumeMonitor` is searching/opening some files on both volumes. That might slow down things, because access to the SD is slow (max 300kB/s). Any idea what is the meaning of the `BDMV/BACKUP/index.bdmv` and `BDMV/index.bdmv` files?

kilrah, 2015-10-27:
Those are standard Blu-ray disc filesystems.

Probably some daemon that checks any connected drive to see if it might be a blu-ray...

dk7xe, 2015-11-03:
i have not had the issues during the past days (Murphy seemed to be on holiday as well ;-)

But today i was able to reproduce during the 2nd connect of the taranis to my PC.
The only difference between connect 2 and connect 1/3 was that Companion was already running when i plugged in the Taranis.. 

sdd is the TARANIS flash
sdc is the SDcard

My guess is that the issue is linked to the last lines of connect 1:
> 2015-11-03T06:00:07.122572+01:00 gppc kernel: [  158.149371] FAT-fs (sdd): unable to read boot sector to mark fs as dirty

System log part below

------------- 1st connect ----------------

> 2015-11-03T05:58:33.471237+01:00 gppc kernel: [   64.498085] usb 3-2: new full-speed USB device number 2 using ohci-pci
2015-11-03T05:58:33.673254+01:00 gppc kernel: [   64.700162] usb 3-2: New USB device found, idVendor=0483, idProduct=5720
2015-11-03T05:58:33.673313+01:00 gppc kernel: [   64.700177] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2015-11-03T05:58:33.673325+01:00 gppc kernel: [   64.700188] usb 3-2: Product: FrSky Taranis Bootloader
2015-11-03T05:58:33.673333+01:00 gppc kernel: [   64.700198] usb 3-2: Manufacturer: FrSky
2015-11-03T05:58:33.673342+01:00 gppc kernel: [   64.700208] usb 3-2: SerialNumber: 00000000001B
2015-11-03T05:58:33.682250+01:00 gppc kernel: [   64.709322] usb-storage 3-2:1.0: USB Mass Storage device detected
2015-11-03T05:58:33.682309+01:00 gppc kernel: [   64.709866] scsi7 : usb-storage 3-2:1.0
2015-11-03T05:58:34.693249+01:00 gppc kernel: [   65.720315] scsi 7:0:0:0: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-11-03T05:58:34.695309+01:00 gppc kernel: [   65.722528] sd 7:0:0:0: Attached scsi generic sg2 type 0
2015-11-03T05:58:34.707345+01:00 gppc kernel: [   65.731244] scsi 7:0:0:1: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-11-03T05:58:34.707416+01:00 gppc kernel: [   65.734304] sd 7:0:0:1: Attached scsi generic sg3 type 0
2015-11-03T05:58:34.711294+01:00 gppc kernel: [   65.738348] sd 7:0:0:0: [sdc] 7862272 512-byte logical blocks: (4.02 GB/3.74 GiB)
2015-11-03T05:58:34.722318+01:00 gppc kernel: [   65.749227] sd 7:0:0:1: [sdd] 1091 512-byte logical blocks: (558 kB/545 KiB)
2015-11-03T05:58:34.729238+01:00 gppc kernel: [   65.756196] sd 7:0:0:0: [sdc] Write Protect is off
2015-11-03T05:58:34.729304+01:00 gppc kernel: [   65.756213] sd 7:0:0:0: [sdc] Mode Sense: 00 00 00 00
2015-11-03T05:58:34.736328+01:00 gppc kernel: [   65.763197] sd 7:0:0:1: [sdd] Write Protect is off
2015-11-03T05:58:34.736391+01:00 gppc kernel: [   65.763210] sd 7:0:0:1: [sdd] Mode Sense: 00 00 00 00
2015-11-03T05:58:34.743261+01:00 gppc kernel: [   65.770197] sd 7:0:0:0: [sdc] Asking for cache data failed
2015-11-03T05:58:34.743318+01:00 gppc kernel: [   65.770208] sd 7:0:0:0: [sdc] Assuming drive cache: write through
2015-11-03T05:58:34.750316+01:00 gppc kernel: [   65.777216] sd 7:0:0:1: [sdd] Asking for cache data failed
2015-11-03T05:58:34.750368+01:00 gppc kernel: [   65.777231] sd 7:0:0:1: [sdd] Assuming drive cache: write through
2015-11-03T05:58:34.834211+01:00 gppc kernel: [   65.861278]  sdc: sdc1
2015-11-03T05:58:34.966230+01:00 gppc kernel: [   65.993277]  sdd:
2015-11-03T05:58:34.984237+01:00 gppc kernel: [   66.011174] sd 7:0:0:0: [sdc] Attached SCSI removable disk
2015-11-03T05:58:35.051335+01:00 gppc kernel: [   66.078197] sd 7:0:0:1: [sdd] Attached SCSI removable disk
2015-11-03T05:58:36.661849+01:00 gppc udisksd[1491]: Mounted /dev/sdd at /run/media/gpeklar/TARANIS on behalf of uid 1000
2015-11-03T05:58:36.673379+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1275]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/index.bdmv
2015-11-03T05:58:36.675012+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1275]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/BACKUP/index.bdmv
2015-11-03T05:58:36.693283+01:00 gppc boot.local[804]: /home: 12.1 GiB (12959186944 bytes) trimmed
2015-11-03T05:58:39.263233+01:00 gppc kernel: [   70.289197] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
2015-11-03T05:58:39.267569+01:00 gppc udisksd[1491]: Mounted /dev/sdc1 at /run/media/gpeklar/OTX_SD on behalf of uid 1000
2015-11-03T05:58:39.329736+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1275]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/OTX_SD/BDMV/index.bdmv
2015-11-03T05:58:39.331245+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1275]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/OTX_SD/BDMV/BACKUP/index.bdmv
2015-11-03T05:59:09.574255+01:00 gppc kernel: [  100.601087] perf interrupt took too long (2511 > 2500), lowering kernel.perf_event_max_sample_rate to 50000
2015-11-03T06:00:07.063232+01:00 gppc kernel: [  158.090662] usb 3-2: USB disconnect, device number 2
2015-11-03T06:00:07.085874+01:00 gppc udisksd[1491]: Cleaning up mount point /run/media/gpeklar/TARANIS (device 8:48 no longer exist)
2015-11-03T06:00:07.122572+01:00 gppc kernel: [  158.149371] FAT-fs (sdd): unable to read boot sector to mark fs as dirty
2015-11-03T06:00:07.126374+01:00 gppc udisksd[1491]: Cleaning up mount point /run/media/gpeklar/OTX_SD (device 8:33 no longer exist)

------------- 2nd connect ----------------

> 2015-11-03T06:00:15.958208+01:00 gppc kernel: [  166.985062] usb 3-2: new full-speed USB device number 3 using ohci-pci
[... cut ...]
2015-11-03T06:03:16.186745+01:00 gppc dbus[808]: [system] Activating via systemd: service name='org.freedesktop.PackageKit' unit='packagekit.service'
2015-11-03T06:03:16.212626+01:00 gppc PackageKit: daemon start
2015-11-03T06:03:16.425180+01:00 gppc dbus[808]: [system] Successfully activated service 'org.freedesktop.PackageKit'
2015-11-03T06:03:34.258772+01:00 gppc PackageKit: get-updates transaction /1_bdbaedad from uid 1000 finished with key-required after 17764ms
2015-11-03T06:03:51.977837+01:00 gppc PackageKit: daemon quit

dk7xe, 2015-11-03:
accidently closed it

projectkk2glider, 2015-11-03:
You have a fs problem on SD card, this looks like it is causing the delay:
```
2015-11-03T05:58:39.263233+01:00 gppc kernel: [ 70.289197] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

The error about `unable to read boot sector to mark fs as dirty` happens after you disconnect the USB cable. Did you first unmount the drives or did you just pull the USB cable?

dk7xe, 2015-11-03:
Disconnect of the Taranis was done without un-mounting the devices before. But my experience in the past was that this didn't make any difference.

What i have done after the 3 tests was replacing the SDcard by a new Samsung 16GB UHC-1.
But after that i had to wake up my kids for going to school ;-) Therefore i will continue my tests tonight/tomorrow morning.

projectkk2glider, 2015-11-03:
You should run fsck on the SD card, assuming it is the same device  as above (/dev/sdc1) do:
```
$ sudo umount /dev/sdc1
$ sudo fsck.vfat -a /dev/sdc1
```

Then disconnect the USB cable and connect it again.

dk7xe, 2015-11-05:
Hi Damjan, i have spend some time on this issue this morning again.
My conclusion so far is that it might not be directly related to a corrupt filesystem on sdc1.

## 1st connect
### Connect sequence:
1) plug in Taranis
2) filemanager opens the two devices connected (both within 5sec).
3) Start Companion
4) click read data --> worked

### Result:
/dev/sdc1 marked bad but reading data worked immedeately

```
2015-11-05T05:56:32.404862+01:00 gppc kernel: [   89.537043] usb 3-2: new full-speed USB device number 2 using ohci-pci
2015-11-05T05:56:32.604916+01:00 gppc kernel: [   89.737148] usb 3-2: New USB device found, idVendor=0483, idProduct=5720
2015-11-05T05:56:32.604969+01:00 gppc kernel: [   89.737159] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
2015-11-05T05:56:32.604981+01:00 gppc kernel: [   89.737166] usb 3-2: Product: FrSky Taranis Bootloader
2015-11-05T05:56:32.604993+01:00 gppc kernel: [   89.737171] usb 3-2: Manufacturer: FrSky
2015-11-05T05:56:32.605000+01:00 gppc kernel: [   89.737177] usb 3-2: SerialNumber: 00000000001B
2015-11-05T05:56:32.613858+01:00 gppc kernel: [   89.746320] usb-storage 3-2:1.0: USB Mass Storage device detected
2015-11-05T05:56:32.613911+01:00 gppc kernel: [   89.746813] scsi7 : usb-storage 3-2:1.0
2015-11-05T05:56:33.625855+01:00 gppc kernel: [   90.758280] scsi 7:0:0:0: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-11-05T05:56:33.626879+01:00 gppc kernel: [   90.759182] sd 7:0:0:0: Attached scsi generic sg2 type 0
2015-11-05T05:56:33.638188+01:00 gppc kernel: [   90.769312] scsi 7:0:0:1: Direct-Access     FrSky    Taranis Radio    1.00 PQ: 0 ANSI: 2
2015-11-05T05:56:33.638250+01:00 gppc kernel: [   90.770597] sd 7:0:0:1: Attached scsi generic sg3 type 0
2015-11-05T05:56:33.643900+01:00 gppc kernel: [   90.776302] sd 7:0:0:0: [sdc] 31257600 512-byte logical blocks: (16.0 GB/14.9 GiB)
2015-11-05T05:56:33.654883+01:00 gppc kernel: [   90.787239] sd 7:0:0:0: [sdc] Write Protect is off
2015-11-05T05:56:33.654940+01:00 gppc kernel: [   90.787256] sd 7:0:0:0: [sdc] Mode Sense: 00 00 00 00
2015-11-05T05:56:33.661918+01:00 gppc kernel: [   90.794285] sd 7:0:0:1: [sdd] 1091 512-byte logical blocks: (558 kB/545 KiB)
2015-11-05T05:56:33.668901+01:00 gppc kernel: [   90.801238] sd 7:0:0:0: [sdc] Asking for cache data failed
2015-11-05T05:56:33.668954+01:00 gppc kernel: [   90.801252] sd 7:0:0:0: [sdc] Assuming drive cache: write through
2015-11-05T05:56:33.675870+01:00 gppc kernel: [   90.808211] sd 7:0:0:1: [sdd] Write Protect is off
2015-11-05T05:56:33.675953+01:00 gppc kernel: [   90.808227] sd 7:0:0:1: [sdd] Mode Sense: 00 00 00 00
2015-11-05T05:56:33.686881+01:00 gppc kernel: [   90.819192] sd 7:0:0:1: [sdd] Asking for cache data failed
2015-11-05T05:56:33.686937+01:00 gppc kernel: [   90.819207] sd 7:0:0:1: [sdd] Assuming drive cache: write through
2015-11-05T05:56:33.757885+01:00 gppc kernel: [   90.890264]  sdc: sdc1
2015-11-05T05:56:33.903868+01:00 gppc kernel: [   91.036265]  sdd:
2015-11-05T05:56:33.910846+01:00 gppc kernel: [   91.043268] sd 7:0:0:0: [sdc] Attached SCSI removable disk
2015-11-05T05:56:33.988921+01:00 gppc kernel: [   91.121269] sd 7:0:0:1: [sdd] Attached SCSI removable disk
2015-11-05T05:56:35.568763+01:00 gppc udisksd[1567]: Mounted /dev/sdd at /run/media/gpeklar/TARANIS on behalf of uid 1000
2015-11-05T05:56:35.590468+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/index.bdmv
2015-11-05T05:56:35.592831+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/BACKUP/index.bdmv
2015-11-05T05:56:38.406917+01:00 gppc kernel: [   95.539343] FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
2015-11-05T05:56:38.414714+01:00 gppc udisksd[1567]: Mounted /dev/sdc1 at /run/media/gpeklar/9016-4EF8 on behalf of uid 1000
2015-11-05T05:56:38.426361+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/9016-4EF8/BDMV/index.bdmv
2015-11-05T05:56:38.428045+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/9016-4EF8/BDMV/BACKUP/index.bdmv

2015-11-05T06:01:49.159891+01:00 gppc udisksd[1567]: Cleaning up mount point /run/media/gpeklar/9016-4EF8 (device 8:33 is not mounted)
2015-11-05T06:05:47.687875+01:00 gppc kernel: [  644.820466] perf interrupt took too long (5009 > 5000), lowering kernel.perf_event_max_sample_rate to 25000
2015-11-05T06:07:19.213213+01:00 gppc udisksd[1567]: Cleaning up mount point /run/media/gpeklar/TARANIS (device 8:48 is not mounted)
2015-11-05T06:07:19.250564+01:00 gppc udisksd[1567]: Unmounted /dev/sdd on behalf of uid 1000
2015-11-05T06:07:19.379931+01:00 gppc kernel: [  736.512403] sdd: detected capacity change from 558592 to 0
2015-11-05T06:07:20.804861+01:00 gppc kernel: [  737.937357] sdc: detected capacity change from 16003891200 to 0
```
## running fsck 
> umount /dev/sdc1
> fsck.vfat -a /dev/sdc1

### Result:
dirty bit was removed and filesystem was ok.

## 2nd connect
### Connect sequence:
1) Companion was kept open since 1st connect 
2) Connect taranis
3) Wait until both devices are showing up under Thunar file manager as devices
4) click read data --> did not work!

### Result:
[... cut ...]
2015-11-05T06:22:27.221774+01:00 gppc udisksd[1567]: Mounted /dev/sdd at /run/media/gpeklar/TARANIS on behalf of uid 1000
2015-11-05T06:22:27.231932+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/index.bdmv
2015-11-05T06:22:27.233577+01:00 gppc org.gtk.Private.UDisks2VolumeMonitor[1338]: libbluray/bdnav/index_parse.c:162: indx_parse(): error opening /run/media/gpeklar/TARANIS/BDMV/BACKUP/index.bdmv
```

projectkk2glider, 2015-11-08:
Based on the data above, your system does not auto-mount both Taranis partitions in all cases. The reason for this is not obvious. Perhaps you can find a similar problem with USB mass storage removable devices on your distribution on some forum, etc..

dk7xe, 2015-11-17:
What i can tell is that Auto-Mount has been working on the 1st connect during this test.

What i have found on https://forums.opensuse.org/showthread.php/489754-USB-storage-not-auto-mounting
is the following statement:
> ..Devices are detected by udev (via kernel events), but removable media mounting has been handled by the desktop environment for a few years now.

Means the issue might be with my xfce desktop. 

What i have done to solve the problem for me until i find the real root cause is to add both product id's to the udev rule. At least on my computer it triggers the auto mount successfully every time.

45-companion-taranis.rules contains now both lines 
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="df11", MODE:="0666";
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5720", MODE:="0666";

I will further investigate if the issue is the same than here --> http://ubuntuforums.org/showthread.php?t=2034756

projectkk2glider, 2015-11-18:
I will close this now as it seems like a one user problem. Please report back if you come up with a better solution.

closed 2015-11-18 by projectkk2glider

dk7xe, 2016-01-05:
The issue is finally solved on my side. 
Under xfce the xfce4-mount-plugin supports mounting file systems into user space.
Installing the openSuse package "xfce4-panel-plugin-mount" solved the problem.

projectkk2glider, 2016-01-05:
Thanks for the update.