opentx/opentx #2988: LUA: inputs range being limited to +-127

athertop, 2015-10-23:
Here's a basic LUA input screen/script which demonstrates a couple of Inputs. In the LUA screen this allows setting of 2 values:

--script start
local inputs = {
	{"Blah 1", VALUE, -200, 200, 0},
	{"Blah 2",   VALUE, -200, 200, 30}
	{"Blah 3",   VALUE, 0, 300, 20}
	}

local function run_func(blah1, blah2, blah3)
--<some code goes here>
end

return {run=run_func, input=inputs}
--script end

The inputs are defined like this: {"label", VALUE, <min_value>, <max_value>, <default_value>}

In my example script above, blah1 should allow the user to set any value between -200 to +200 with the default value being 0. It does not however! It allows only the values -128 to +127 (so only 256 distinct values being allowed).

In the same script, blah2 should allow the user to set any value between -200 and +200 with the default value of +30. It does not however! It allows the user to set any value between -98 and +157. (so the default seems to be getting used as a range offset but retaining the 256 maximum distinct allowed values)

In the same script, blah3 should allow the user to set any value between 0 and +300 with the default value of +20. It does not however! It allows the user to set any value between 0 and +147. (the offset of 20 being applied here again to values. The min remains at 0 whilst the upper limit is offset by 20 to 147)

So to sum up - the pattern of value definition vs. what is seen in practise shows that there are a maximum of only 256 distinct values which we are allowed to set for any given input, and the centre point of the range of values is offset by the default value specified.

If this is intentional, then it needs to be documented, if not, then can this be fixed so we can have some reasonable range of input values available to use in scripts?

Thanks, Paul

dsbeach, 2015-10-25:
I'm looking into this now.  The documentation will be updated to reflect a valid range of -128 thru +127.

However, if an input is described as {"in2", VALUE, -128, 127, 30}, the behavior is still not as expected.  In that case the maximum of 127 is enforced, but when reducing the input value via the minus key it goes from -98 to +127.

I'll be looking into the code today.

dsbeach, 2015-10-25:
Reproduced in simulator for 2.0 and 2.1. Tagged as very low priority bug.

athertop, 2015-10-25:
Where you say:
However, if an input is described as {"in2", VALUE, -128, 127, 30}, the behavior is still not as expected. In that case the maximum of 127 is enforced, but when reducing the input value via the minus key it goes from -98 to +127.
I agree with your findings, although if you set the upper limit to say 200, then the range selectable (with +- keys) becomes -98 to 157.

Is it possible to code openTx to allow a wider range of values? I use several lua screens for entering values used by telemetry lua scripts and having 256 possible numbers is extremely limiting (means a lot of creative coding required to make use of that mechanism). Having a wider range of values would be of real benefit for coders wishing to extend the functionality of their OpenTx radios.

dsbeach, 2015-10-25:
Paul,

I'm just getting started with OpenTX development.  My earlier comments are
based on what I assume was the intended limits based on the behavior I've
observed. Using limits beyond the intended range are likely to cause
unpredictable results.

Can you provide a use case for the wider range?  There may be an easy way
to accomplish your intended use.

athertop, 2015-10-25:
dsbeach, Thanks for your responses - most appreciated.

As for a use case, I am trying to assist (with tesing) the guys who are developing an awesome lua script for Pixhawk-FrSky telemetry. The link to the overview wiki is here: https://github.com/Clooney82/MavLink_FrSkySPort/wiki/0.-Overview

This script relies on the user setting parameters in a couple of lua scripts (manually added to the lua page). The sorts of parameters entered are for things like mAh (LiPo Current capacity), Wh (Lipo power Capacity), Warning/Critical cell voltage values. Here are some screenshots of these pages:
https://github.com/Clooney82/MavLink_FrSkySPort/wiki/2.1.1---Audio-Message-Minimum-Cell-Info
https://github.com/Clooney82/MavLink_FrSkySPort/wiki/2.1.2-mAh-and-Wh-telemetry-screen-meter-calibration.
So as you can see, having only -128 to 127 as possible values is rather limiting. I think by having a wider range of possible selection values, plus also maybe having the option for accuracy (decimal places - as is the case on the new telemetry sensor setups) would provide a platform for more advanced lua development, potentially making it more popular, and this certainly can't halm the benefits list of RC enthusiasts moving to the OpenTX platforms. Have I sold it enough yet?? ;-)

kilrah, 2015-10-25:
IMO the way these scripts use to set the parameters is not really appropriate, on 2.1 it should rather be done as one (or 2, with one being for settings) telemetry scripts, with settings stored on the SD card.

projectkk2glider, 2015-10-25:
Its a bug: everything in lua_api.cpp is written using int16_t for value, min and max. But the actual value selected by the user is stored in int8_t variable in structure g_model.scriptsData[].inputs[].

@bsongis What do we do:
* increase size of `ScriptData`, needs EEPROM change, there is no spare place there.
* or we limit it to int8_t in Lua altogether.

bsongis, 2015-10-25:
Wouldn't it possible to use 2 8-bits inputs as 1 16-bits for this kind of script? It would avoid an EEPROM change

Clooney82, 2015-10-25:
@kilrah: How can I store settings on SD-Card and reread them?
I also tried to put some mixer scripts together, but this didn´t worked well, cause not all inputs are shown in script.
How many input values are possible in mixer scripts?

projectkk2glider, 2015-10-25:
>How many input values are possible in mixer scripts?

10 per script.

kilrah, 2015-10-25:
> @kilrah: How can I store settings on SD-Card and reread them?

Store them in a file you access using the lua io library.
http://www.lua.org/manual/5.1/manual.html#5.7

dsbeach, 2015-10-26:
I've spent some time stepping through the code, and have decided I'm not comfortable attempting to implement a code change. I will update the new documentation to reflect the limit of 8 inputs (and 8 outputs?)

dsbeach, 2015-10-26:
Through testing I have determined that outputs are limited to 6 which will be reflected in the documentation.

mpaperno, 2017-11-29:
Input limits are now int16 size in v2.2.

closed 2017-11-29 by mpaperno