opentx/opentx #2991: Timer as input not possible at 2.1.3 and Nightly

janekx, 2015-10-25:
On 2.0 was possible to set timers to inputs. On 2.1.3 and nightlyes is not possible to select it in the input. 
I tried to set timer, give it name but there in the selectons of inputs no longer appear.

projectkk2glider, 2015-10-25:
On the radio or in Companion?

janekx, 2015-10-25:
Sorry i tried it on radio

kilrah, 2015-10-25:
Inputs have never been a possible choice for throttle source.

janekx, 2015-11-01:
Talking about that is possible to use timer as progress to the any channel. 
https://www.youtube.com/watch?v=rEs3y2xj7xE

kilrah, 2015-11-01:
Ah, makes more sense. Not sure if it's a bug or intended.

janekx, 2015-11-01:
It is unususal function but for some like tanks or other it is useful, that is what makes the OpenTX that OpenTX where can be done anythink :)

kilrah, 2015-11-02:
I had a look and it was apparently removed because some code it needed to work is no more part of the new telemetry system.
I.e. quite a bit of work to put back in, so unless there are "real" scenarios where it's useful it will likely stay as it is.

projectkk2glider, 2015-11-02:
@janekx  I suggest to use a Lua script as an alternative. I should be fairly simple and could also do much more if needed.

closed 2020-07-07 by stale[bot]