opentx/opentx #2994: [Bug][Radio][OpenTX 2.1.3] All "Trainer port" options are not available

LapinFou, 2015-10-25:
Hi folks,

I installed a X8R in the JR bay in order to setup a wireless trainer link. I discovered 2 issues:
1- The "Start" & "Channels" fields are missing in the radio firmware, but are present in Companion
[image]

2- The "Master/SBus in battery compartment" is available under Companion, but not in the radio.
[image]

By the way, for this kind of setup, what is the most appropriated failsafe setup for the student radio?
The current setup I'm using is: D16 / Ch: 1-16 / Failsafe mode: No Pulses

kilrah, 2015-10-25:
The issue is wrong, the mistake is that they should not be visible in companion.

For the failsafe you should have nothing to configure when using sbus. Try it.

closed 2015-10-25 by kilrah

LapinFou, 2015-10-25:
Hi Kilrah,

I didn't understand your answer.
The master radio has got a X8R (binded with the student radio) inside the JR bay. A wire is plugged between the S.Bus port of the X8R and the JR pins.
How the master radio will detect if the link between the student radio and the X8R (used as a wireless trainer port) is lost?
When I was taking about the current setup, I was talking about the internal RF setup of the student radio. Thinking that "no pulses" was the correct setting to help the master radio to detect a lost of connexion. 

What about the 2nd point: "The "Master/SBus in battery compartment" is available under Companion, but not in the radio."?
Is a Companion or radio bug?

bsongis, 2015-10-25:
The scrollbar part of the issue was fixed earlier today. It's available in the last 2 "forced" nightly builds

bsongis, 2015-10-25:
Master/SBus in batt compartment is available only if you have it available in your Hardware menu

LapinFou, 2015-10-25:
Thanks !! I missed this one. Then, it means Companion doesn't take in account this settings. However the simulator does (I checked it).

kilrah, 2015-10-26:
> How the master radio will detect if the link between the student radio and the X8R (used as a wireless trainer port) is lost?

The SBUS stream includes a "Failsafe" flag that the receiver will set if it has no link, and that the master radio will use to declare the input lost. This happens regardless of any failsafe setting on the slave radio. When using SBUS, failsafe settings only matter for devices that are not making use of that flag.

LapinFou, 2015-10-26:
Thanks for the explanation. I didn't know that the S.Bus protocol did included a failsafe flag.
Last question, concerning the failsafe flag, is it the same with the CPPM protocol?

kilrah, 2015-11-01:
No, PPM can not transmit any info besides channel timings.

commit 01226b61 (2015-10-25): No PPM options are available if trainer mode is master (fix #2994)