opentx/opentx #2997: [Feature Request] Trainer port connect/disconnect indicator

LapinFou, 2015-10-26:
Hi folks,

Since some people are doing some complex training setup (the most common one is: the master radio will regain the control by moving its sticks), it would be very nice to have "trainer port connect/disconnect" information.
The built-in trainer mechanism will switch back to the master controls if the trainer port is detected as disconnected. This is nice and safe.
If you create a complex setup using the TR1-TR16, this is not the case. The easy solution would be to create a logical switch to detect if the trainer port is present or not. Unfortunately this information is not available.
Is it possible to add in the list a **"Trainer is present"** parameter?

This should work with DSC port, S.Bus/CPPM in the JR bay and S.Bus int he battery compartment.

kilrah, 2015-10-26:
Any TRxx mixer line gets disabled automatically when trainer is disconnected.

closed 2015-10-26 by kilrah

LapinFou, 2015-10-26:
Well, this is not the case in my current setup (X8R in the JR bay/S.Bus wired). If I switched off the student radio, I got a voice announcement for the trainer KO, but TRx takes a value depending of the failsafe configuration of the X8R.
I will do more testing at home to be more accurate, but your statement doesn't match my observations.

kilrah, 2015-10-26:
Then that would be a bug that needs solving, not something that needs a new feature.

LapinFou, 2015-10-26:
OK. I will do more testing ASAP.
However, having a trainer state indicator would be nice.
:smiley:

LapinFou, 2015-10-26:
I was thinking about something: without a "trainer mode active" indicator, how can we use more than 4 channels? To me, this indicator is really missing. I must think about more arguments to convince the team that it would be a good idea to add it.
:smiling_imp: