opentx/opentx #2999: [Taranis] Problem with SBUS Trainer mode

bsongis, 2015-10-27:
Reported by Wayne: "we had a few reports from people. We did most all of our testing with the D4R-II in CPPM mode. Only ground checks with Sbus, so we failed to fully test it. Sorry!! The issues take a little time to show up, but sure enough it seems to do it every time. 
We also wired in an X8R just to make sure it was not an RF thing, and it had the same results.
from an observation emailed to me this morning:
But the worst part was that after several minutes with the
Student flying, the plane would go crazy, and returning control
to the Instructor it started working OK."

But it means ... who has the RSSI warning?

"I think the master radio does. I was out sick this weekend, so was not present for this round of testing. (very sick..)"

would you double check if you can reproduce?
I just read again the code, I really see nothing, I hope it's not just interferences between the 2 modules

"I just asked Bruce a few more details from his weekend testing. He was doing the additional testing as we had heard from a few people about issues when using Sbus with the Linker. It may be a while before I have an answer for you."

kilrah, 2015-10-27:
Another report:
http://www.rcgroups.com/forums/showpost.php?p=33039291&postcount=14

And in French:
http://frskytaranis.forumactif.org/t2828-rx-sans-telemetrie-avec-sortir-cppm#33090

kilrah, 2015-10-27:
If we could have details on what firmware the receivers are using (EU/non-EU) it would be nice too to see if that can play a role.

bsongis, 2015-10-27:
Here is an update from Bruce:
**I could never get the bad RSSI Critical/Telemetry Lost/Recovered messages
on the ground.
Maybe the model receiver needs to be quite a ways away before it happens?

But I had some success seeing the the XR4-SB Trainer channels misbehave.

I used a D4R-II as the model receiver and used channels 1-4 for the Trainer.

With the Trainer Radio, Student Radio, and Model receiver all about 20 feet apart
I switched a few times between Student and Trainer, then kept it
with Student in control.

After a few minutes, looking at the Channels Monitor, every now and then
I could see the 4 channel values jump around.

Sometimes you had to wait several minutes for it to happen, then it would
happen for maybe 5 seconds.
**

kilrah, 2015-10-27:
> I could never get the bad RSSI Critical/Telemetry Lost/Recovered messages
on the ground.
Maybe the model receiver needs to be quite a ways away before it happens?

This would be "normal", the X receiver having its antennas right next to the radio's module it can swamp it and prevent it form receiving the telemetry from the model when it is far. Moving the antennas further away, even a few cm will help.

> After a few minutes, looking at the Channels Monitor, every now and then
I could see the 4 channel values jump around.
Sometimes you had to wait several minutes for it to happen, then it would
happen for maybe 5 seconds.

For this the only thing I can imagine is the opposite, with the master radio's module preventing the X8R from receiving control data from the slave and causing it to enter/exit failsafe mode and cause the jumps.

Both systems send RF at a 9ms interval, but with imprecisions both timings will slowly "slide" with regards for each other, and sometimes timing might be "just right" to block the trainer RX.

If I'm right then nothing but physical separation can solve it.

AloftHobbies, 2015-10-27:
This would be with the "normal" X based stuff.

We can give Kilrah's separation idea a quick test. I think I heard of someone having the same issue when using Sbus into the trainer port (DSC). I think they may have been running a different brand RF system. I may be wrong on that.. I speed read a lot of different stuff for the FrSky world.

CPPM does not seem to have an issue. Also, we will have the X4R-SB-CPPM receivers in very soon, should be interesting to give one of those a try in CPPM mode and see if they still have the same issue, as the RF stays the same that way.

kilrah, 2015-10-27:
> I think I heard of someone having the same issue when using Sbus into the trainer port (DSC). I 

I doubt since that doesn't work! SBUS can only come in either on module pin 2 or on the battery bay serial port.

AloftHobbies, 2015-10-27:
LOL - See, my memory...

Ze-baloo, 2015-10-28:
I will add my observation, sorry in advance for my lack of electronical knowledge ;-)

I have binded a X8R to my Turnigy 9X under OpenTX-9X 2.1.3 with a stock frsky DHT module.
The X8R has been connected in the external module bay of my Taranis X9D A01 model under OpenTX 2.1.3 (see photo)
[image]
All are on non-Eu firmware.

While observing the Chanel monitor on my Taranis once the trainer activated, i have noticed some "phantom" slave commands. The problem is that it happens all the tims independantly of the model used, more annoying, it will happen, but no clue when, i mean that it's not clearly reproductible.

if you want more information, please ask, it'll be a pleasure to help you to find out.

LapinFou, 2015-10-28:
I have the same setup as Ze-baloo. I also got some troubles.
In my setup the X8R is binded in mode 2 (D16 / 16ch / telemetry off).

banderson546, 2015-10-28:
I'm testing a Linker equipped with an X4R-SB on 2.1.3 with 4 channels set for the Trainer function activated by a switch.

I did an experiment and separated it from the Taranis by 3 feet using an extension cable.

It still got the occasional channel glitching coming over the link from the Student receiver.
There were no announcements of Trainer signal loss.
It was very intermittent, sometimes it would take a while to happen. It usually would last for 3 or 4
seconds and then stop.

KIrill-ka, 2016-01-25:
@kilrah, do you confirm that issues mentioned here are caused by uart interrupt latency (overrun errors) as in #3252?

kilrah, 2016-01-25:
I can't confirm it as it wasn't investigated yet. If you did and can affirm it then great!

KIrill-ka, 2016-01-25:
I was a bit worried that my "investigation" was put aside :)
It's not absolutely clear to me that this is an exact duplicate of #3252.
By the way, as a temporary solution I shortened sbus frame which is sent to taranis to 23 bytes. Now there's no glitches. That's because OREs make frame too short to be accepted.

kilrah, 2016-04-13:
@bsongis Did we change to DMA on master in the end (worth restesting this)?

banderson546, 2016-04-13:
I've retested this on 2.1.7 and I still see see channels glitching over the trainer link.  It takes
a minute or two before it happens and it lasts for several seconds.

kilrah, 2016-04-13:
2.1.7 doesn't include any fix for that, it would be on the 2.1.8 nightly. But I jsut had a look and it doesn't seem anything was done on 2.1 yet.

LapinFou, 2016-05-03:
OTX 2.1.8 is out, but this bug has not been fixed.
Are you planning to fix it for the milestone 2.1.9?

banderson546, 2016-05-07:
On the same Taranis that generated glitches with OpenTX, I flashed ErSky9X and configured it as an SBUS Trainer.

I did not get any channel  glitching with ErSky9X.

bsongis, 2016-05-07:
Would you check 2.2? 
Here:
http://downloads-22.open-tx.org/nightly/companion/
And you enable the firmware nightly builds in the Settings to download the latest firmware

banderson546, 2016-05-07:
I've tried with 2.2.0N300 and SBUS Trainer appears to be broken.  I activate Trainer in Special Functions and control just stays with the Trainer, there is no control from the Student Radio.

AloftHobbies, 2016-05-09:
Bertrand thanks for digging into this. Would be great to have a fix with the next release as we are looking to expand on the Sbus functions. :)

bsongis, 2016-05-09:
Yes but this is fixed in 2.2, I just need somebody gives a test, if it works, I will port it to 2.1 (next stable release)

projectkk2glider, 2016-05-09:
@banderson546 says that 2.2 does not work!

bsongis, 2016-05-09:
Sorry you are right, I will check later tonight

projectkk2glider, 2016-05-09:
The cause for this could be:
* we get UART overrun in the middle of SBUS frame (confirmed by #3252)
* we still process such SBUS frame even if it is only 24 bytes long (normal SBUS frame is 25 bytes)
* this frame will have all channels from the missing byte and up pretty well corrupted.

Now, the reason for the buffer overrun:
* on `master`: #3252 says that disabling internal module cures the problem, so this interrupt is taking too much time (occasionally). Remember SBUS speed is 100kBps, character time is 100us.
* on `next` we receive trough DMA so there should be no overrun. Must be some other reason, one could be the issue of `DMAFifo.isEmpty()` in relation to DMA NDTR register. 

Question for @bsongis why do we allow SBUS frames with a length different from 25 bytes?

bsongis, 2016-05-09:
It should work the same way than the serial port UART3 or S.PORT on Horus!
I am afraid of a collision with external module pulses, as it also uses DMA2_Stream2

bsongis, 2016-05-10:
Instead I am afraid we never detect the end of the frame with the pop() done in the mixer task, the SBUS fifo is perhaps never empty with the mixer running once every 20ms!

KIrill-ka, 2016-05-11:
@bsonis, As a temporary fix I changed s.bus source to send truncated 23 byte frames. It did help with channel glitch - bad frames were ignored.

bsongis, 2016-05-12:
On 2.1?

bsongis, 2016-05-12:
I need to discuss with @projectkk2glider about the best way to fix it in 2.2

projectkk2glider, 2016-05-12:
My proposals:
* use DMA2_Stream1 for heartbeat input, this will prevent collision with  DMA2_Stream2.
* call `processSbusInput()` directly in mixers thread every time it is run (every 2ms)
* change the SBUS fame synchronization in `processSbusInput()`, wait for start character, data bytes and valid end byte. Then parse SBUS frame and have more strict checks (especially the length which should always be 25 bytes).

KIrill-ka, 2016-05-13:
> On 2.1? 
Yes. 2.1.7

bsongis, 2016-05-13:
I just moved the call to processSbusInput so that it is called every 2ms and it seems to work perfectly now

bsongis, 2016-05-13:
[image]

bsongis, 2016-05-13:
The delay is far greater than 2ms, so it's normal it works. I will wait for more tests in 2.2 then I will port the DMA thing to 2.1

projectkk2glider, 2016-05-14:
I am not completely satisfied. Two more things to do:
* use DMA2_Stream1 for heartbeat input, this will prevent collision with DMA2_Stream2.
* have more strict checks (especially the length which should always be 25 bytes).

bsongis, 2016-05-14:
Who wants to give some tests before we merge it into 'master'?

projectkk2glider, 2016-05-14:
We could still have glitches, I think we should only accept SBUS frames with 25 bytes, not 23 bytes as it is now. This is what actually produced weird channel errors when one byte was missing from the SBUS frame. Why do we allow 23 bytes anyway?

banderson546, 2016-05-15:
I'll give it a test. I don't see it yet though in the 2.2 nightly's.

raphaelcoeffic, 2016-05-16:
@projectkk2glider I remember wondering about the same as I looked at the code last year. I double checked just now in librepilot & cleanflight. Librepilot checks for the same things you proposed, whereby cleanflight does not check the end-byte.

banderson546, 2016-05-19:
Thanks Bertrand!
I tested 2.2.0N317 and it is working perfectly on the ground now :)
I will do more testing in the air this weekend.

AloftHobbies, 2016-05-19:
Yay!! Great to hear, especially if we want to do some more cool things with Sbus. :)

LapinFou, 2016-05-22:
2 questions:
1. I assume the port in the battery compartment is also impacted by this bug and the fix will also applied to it. Correct?
2. I'm planning to setup a wireless body box on my T+ using the (UART?) port in the battery compartment. Is it possible to achieve the S.Bus trainer mode **AND** used the Spektrum hack module in the JR bay module at the same time?

bsongis, 2016-05-22:
1. Yes the serial port in the battery compartment is also impacted, it now uses DMA in 2.2 as well.
2. Yes it should work until you open a new issue

LapinFou, 2016-05-22:
@bsongis 
Thanks !! :smile:

banderson546, 2016-05-23:
I've tested 2.2.0N317 in the air and it does not
have issues with channels glitching now.

It still has a problem though with the Telemetry
link between the Trainer Radio and the model.

Several times a minute the Trainer Radio announces Telemetry 
Lost-Telemetry Recovered and the Telemetry display
disappears and then comes back.
It only happens when the SBUS channel is up.
If the Student Radio is turned off Telemetry
returns to normal.

bsongis, 2016-05-23:
Are you sure this alarm comes from the Trainer radio? Are the 2 radios close each other? Is the model close to the Trainer radio?

kilrah, 2016-05-23:
Try disconnecting only the SBUS wire between the receiver and the master radio, i.e. leaving the receiver powered.
The telemetry lost/recovered is most likely due to the receiver swamping the telemetry reception when it sends its own back, i.e. nothing to do with software, but something that would be solved by physically putting the receiver further away.

LapinFou, 2016-05-23:
I'm using the same setup of @Ze-baloo with a SBus to CPPM converter (until the issue is fixed).
I was getting lots of telemetry lost/recovered voice message. So I bind the RX in mode 2 (D16 without telemetry). This solved the problem. As Kilrah said, I'm pretty sure the TX part of the receiver is much too close of the internal XJT antenna and create some kind of saturation.

AloftHobbies, 2016-05-23:
That is odd. If it was swamping, it would be the student's radio that would have the telemetry lost errors, not the instructors. I have never found a second radio or receiver being close to my radio would cause telemetry lost, but I will go test that right now...

Yep just tested and confirmed. It would appear that only the bound receiver being too close causes the swamping issue.. This makes no sense to me, but I thought that might be the case. **Begs the question if something else is going on with the swamping issue.**

banderson546, 2016-05-24:
I used an extension cable to move the Linker 3 feet away from the Trainer Radio and the 
Telemetry Lost problem with the model does go away. 

I don't see a way to disable telemetry on an X4RSB receiver though.

kilrah, 2016-05-25:
> That is odd. If it was swamping, it would be the student's radio that would have the telemetry lost errors, not the instructors. 

Nope, the instructor's is the one that has 2 transceivers basically an inch away from each other which is the whole problem. The student radio only has one.

The same problem also sometimes happens when you use both internal and external module together.

LapinFou, 2016-06-08:
@AloftHobbies 
Hi Wayne,
It would be very nice if you could request to FrSky a firmware for the X4R where the telemetry can be disabled. Otherwise, you will get some issues with your "Linker" product when using a XR4-SB.

AloftHobbies, 2016-06-08:
Limited testing here (with dead batteries) did not cause this telemetry lost issue.

I would like to test using Mike's firmware and see if we have the same issue then. I am not convinced we have pinned this down yet.

ledvinap, 2016-08-02:
There is possible problem with some sbus receivers - heartbeat line is connected both to internal and external module. Internal module on Taranis X9D has open-drain output + small resistor in heartbeat signal.
With X8R this works fine, receiver drive strength is large enough to fight heartbeat from internal module (signal is pulled down by about 0.5V).
But with R9D SBUS signal is pulled down considerably(to ~1.5V), at least under threshold of saleae logic analyzer.
This problem is more pronounced in D8 mode, heartbeat is pulled low much longer than in D16 / LR12.

ledvinap, 2016-08-02:
Another problem with Radiolink X9D (after cutting internal module heartbeat):
Trainer signal returns to zero(low value) for short time (probably one tick, almost always visible on channel display, usually registered by servo).
The pattern is very stable, is is 3 glitches about 1s apart, repeating every 8 seconds.
X9D is in failsafe mode, data are stable (all packets are identical). SBUS packets are send every 13.44ms, delay between packets is 10.48ms, with jitter less than 0.2ms
The frames (failsafe) are:
F0 37 D8 F9 EE 2B E3 36 66 C8 F6 04 0B E0 00 D0 9A 10 02 00 40 08 01 08 00
The glitch happens both with and without failsafe bit set. This is a bit strange, failsafe SBUS packets should be ignored ..

This does not happen when I connect FrSky X8R .. 

Any idea what can be causing this?

Using version 2.1.8

ledvinap, 2016-08-02:
Ok, the failsafe not triggering is probably caused by X9D sending wrong bit.

ledvinap, 2016-08-03:
Bisected problem to  #3178. SBUS overrun + ignoring serial port errors is probably causing my problem ...

sbus trainer works fine with https://github.com/opentx/opentx/pull/3410 applied

AloftHobbies, 2016-09-12:
So the trainer feature works for you now? Regardless of receiver?

ledvinap, 2016-09-12:
@AloftHobbies : 
It does work with #3410 applied and heartbeat from internal module disconnected.
It may work with heartbeat in original state, I did not test that.

Radiolink X9D receiver failsafe/flag byte is reversed, so 'trainer signal lost' is not reported, but that is not OpenTx problem (I modified it to accept both bits as failsafe, but that's only hack)

[29 more replies]

commit 5b7526c5 (2016-05-13): Fixes #2999 in 'next' (OpenTX 2.2)
closed 2016-09-14 by bsongis