opentx/opentx #3000: 1 of 4 related issues to LUA io library

sbright33, 2015-10-27:
How do I see the Error on the Custom Script page?  I can see there is an error, but what it is?  Syntax or run time?  What line?  Please educate me.

sbright33, 2015-10-27:
Is it different or possible in Companion or Taranis?

projectkk2glider, 2015-10-27:
You should see the error in the Companion radio simulator, just press F6 key when you are in the simulator. That will open the *Debug output* window where among other stuff you should see the error message. You will also see the output of any Lua `print()` statements there.

Another more difficult option is to have a special build of OpenTX on the radio that has debug mode enabled. With this kind of firmware you will get similar output to the *Debug output* in Companion. The output is available in two ways (again dependent on the options used to compile OpenTX):
 * serial port in battery bay
 * USB emulated serial port device.

But for that you have to compile the OpenTX yourself with the appropriate compile time options. Are you familiar with the firmware build process?

projectkk2glider, 2015-10-29:
@sbright33 Here something it did not know, you can select firmware option `cli` in the Companion. With that option selected you can use the USB port see Lua errors and other output. Steps:
* flash the radio with firmware that has `cli` option selected
* turn on the radio
* connect to PC via USB for windows, you will need serial port driver from ST: http://www.st.com/web/en/catalog/tools/PF257938
* on PC open terminal  emulation program (it used to be called hyperterminal, I don't know which program has the same functionality now) and connect to the COM port that was created by the radio, use any baud rate.
* press Enter key, you should get the CLI prompt `>`
* type `help` to see the available commands
* type `trace on` to enable debug output (this will enable you to see Lua print() statements)
* disconnect USB cable after you are done.

One small warning: the OpenTX firmware with the `cli` option has less memory available to Lua scripts. I also has additional debugging code included, and that code is not as well tested as the rest of OpenTX code. So I advise caution in a sense that this kind of firmware is best used for testing and not for (expensive) model control.

closed 2015-10-29 by projectkk2glider

kilrah, 2015-10-29:
> on PC open terminal emulation program (it used to be called hyperterminal, I don't know which program has the same functionality now) 

PuTTY works well.

> connect to PC via USB for windows, you will need serial port driver from ST

As I alerady mentioned this version 1.4.0 of the driver seems not to work with the Taranis implementation. Looking for alternatives I found V 1.3.1 that works.

projectkk2glider, 2015-10-29:
>As I alerady mentioned this version 1.4.0 of the driver seems not to work with the Taranis implementation. Looking for alternatives I found V 1.3.1 that works.

Can you post a link to it.

kilrah, 2015-10-29:
Lost the link but I'll put the file in the OpenTX downloads. Tomorrow, too late now.

kilrah, 2015-10-30:
Added. http://www.open-tx.org/2015/10/30/st-vcp-1.3.1-for-taranis-cli/

SandroGrzicic, 2017-03-18:
@projectkk2glider @kilrah can you please confirm that this method still works with OpenTX 2.2? I am trying to debug a LUA script but I cannot get the CLI to work over USB.

I am following the steps above exactly, but when I connect to my X9D+ with Putty, all I get is an empty nonresponsive terminal screen. I am using the 1.3.1 STM32 driver and am on Windows7.

I am using the firmware "opentx-x9d+-noheli-lua-luac-cli-sqt5font-en-2.2.0N364".

Thanks!

kilrah, 2017-03-18:
I believe you need to build with DEBUG=YES which isn't enabled by the cli option i.e. you have to build yourself.

SandroGrzicic, 2017-03-18:
I thought as much, I just spent an hour trying to set up my build environment using CLion. I am nearly done I think, just need to build PyQt4 from source (they discontinued the Windows installer apparently) for which I need to install MS Visual Studio.. I guess I'll finish the process and do a local build with DEBUG=YES. Thanks.