opentx/opentx #3003: 4 of 4 related issues to LUA io library

sbright33, 2015-10-27:
io.write(fh,"any string") is OK
io.write(fh,stringvarname) is OK
io.write(fh,numvarname) writes .14g to the file
io.write(fh,2) writes .14g
io.write(fh,2.2) writes .14g

This is only on Taranis.  All good on PC sim.  That's how I know my syntax is correct?

projectkk2glider, 2015-10-27:
Can you post the whole Lua script as an example that demonstrates all three issues you have: opening file, writing to it and closing. Tnx.

projectkk2glider, 2015-10-27:
Confirmed, the problem is that Lua in this case uses f_printf() with format string ".14g" which is not supported by ff.c.

sbright33, 2015-10-28:
So it's ready for me to test?  How do I download this version you fixed?  Sorry, I'm new to Github, and never built this project.  Can you make it easy for me?

projectkk2glider, 2015-10-28:
The fixes are not yet in main branch, so they are not available in nightly builds. I can prepare a custom build for you, but for that I need to know which radio you have: Taranis, Taranis Plus or X9E? And which language?

sbright33, 2015-10-29:
T Plus. English.  Thanks!

projectkk2glider, 2015-10-29:
Here is your bin file: http://downloads-21.open-tx.org/test/opentx-taranisplus-lua-en-2.1.5-io.bin

Here is my small standalone script (execute from the SD browser):
```lua
local function run(event)
  print("c1")
  local f = io.open("foo.bar", "a")
  io.write(f, "first line\r\nsecond line\r\n")
  io.write(f, 4, "\r\n", 35.6778, "\r\n")  -- one can write multiple items at the same time
  local foo = -4.45
  io.write(f, foo, "\r\n")
  io.close(f)
  return 1    -- this will end the script execution
end

return { run=run }
```

Remember that you are working with unofficial OpenTX release, so use it with caution preferably not with any expensive model. Also make a backup of your EEPROM before flashing (just in case something goes wrong).

The Lua io library is somewhat different from the official version, you can find preliminary docs here: https://github.com/opentx/lua-reference-guide/issues/3

Report back on your findings.

PR #3011 (2015-10-31): Projectkk2glider/lua io issues
I have tested this on my radio (Taranis) and in simu. Both work. The test script:
```lua
local function run(event)
  print("c1")
  local f = io.open("foo.bar", "a")
  io.write(f, "first line\r\nsecond line\r\n")
  io.write(f, 4, "\r\n", 35.6778, "\r\n")  -- one can write multiple items at the same time
  local foo = -4.45
  io.write(f, foo, "\r\n")
  io.close(f)
  return 1    -- this will end the script execution
end

return { run=run }
```

Perhaps we should wait for @sbright33 to test this, the result of his test is expected in #3003.

Closes #3003 and closes # [... cut]
commit 2e068044 (2015-10-31): Fixes #3003: Lua io.write() did not support numbers
closed 2015-10-31 by bsongis