opentx/opentx #3006: Trim for input (using RS) is OFF on radio and ON in companion

griebd, 2015-10-28:
Radio: Taranis Plus
OpenTX: 2.1.4 (self compiled at commit 21a67b8 with HELI=YES GVARS=YES MIXERS_MONITOR=YES TEMPLATES=NO)
I was comparing two models (using companion compare) and noticed that "trim" was OFF on the base model (build in companion) and ON on the radio model backup... turned my radio on and went to fix it BUT on the radio it was OFF! Just to be sure I turned it ON, set it to some other INPUT, and then back OFF. Did a new backup and when opening it in companion, it was back ON. Tried some more and always the same behavior.
Apparently it's something related to the fact that I'm using RS (with obviously has no won trim) in this input. For "normal" inputs that have trim, the OFF persists from the radio backup to companion.

griebd, 2015-10-28:
Just to make it clear (as my explanation is a bit messy) here it's how to reproduce:
1. open companion and File -> New
2. hit Simulate Tx
3. go to INPUTS page and edit the first input
4. change source to Pots -> RS
5. check that trim is OFF
6. exit and backup this eeprom
7. open this backup in companion and check that the edited input has Include Trim = Yes
Hope it's a bit clear now.

projectkk2glider, 2015-10-28:
I guess that trim is set to the default value, since it has no effect whatsoever. The default is yes.

griebd, 2015-10-28:
I don't see a use for it, but if the option is there it should be consistent... if it has no effect it should be disabled! But one can set the rudder trim to change it, does it work?
Anyway, if you set it in companion, open the simulator, do the backup and open this backup on companion, then it's disabled! So, depending on how you do it (radio or companion) you end up with a different eeprom for the same configuration...