opentx/opentx #3008: 9XR128 - baudrate option gone from SETUP

flegmatoid, 2015-10-28:
OpenTX-2.1.3 seems to have Baudrate option missing in "RADIO SETUP" menu (FrSky and Ardupilot build options).

Has it been moved to another location, or simply is gone, or is baudrate detected automatically now?
Didn't find this in the changelogs

kilrah, 2015-10-28:
There's never been a baudrate option for FrSky.
Maybe for mavlink, but I don't understand how you can say it isn't there anymore if it's broken and won't build in the first place (your issue #3009).

flegmatoid, 2015-10-29:
Can this be missing for Ardupilot too?Because I'm nearly confident that was my build, which has lost baudrate option after upgrading from 2.1.0 to 2.1.3. Therefore, I've tried FrSky and Mavlink just to see if there's such an option to, which was not for FrSky (you claim it was never there, which is fine). Would be nice to have this tested/confirmed for Mavlink too, which I cannot do as per #3009

kilrah, 2015-10-29:
Hmm you got me confused for a moment, and it turns out it's the first time I notice there are both a "mavlink" and "ardupilot" version. I don't even know what the difference is, I thought ardupilot used mavlink and they would thus be the same thing in OpenTX...

None of the current developers has much knowledge or need for those nor anything to test them, so they unfortunately tend to get broken. We haven't seen the one who did the integration in the 9x version since a couple of years ago, and that's mostly why it isn't present on newer platforms nor well maintained.

closed 2019-08-19 by kilrah