opentx/opentx #3012: [Companion] Simulator not Working

trashmaster9000, 2015-10-29:
Hi,

I changed things in the Telemetry Tab and now i cant Simulate my Firmware.
I added 3 Sensors (A1, A2, RSSI), changed RSSI Alarms and Added "Numbers" to "Telemetry Screen 1"

Changing back the Telemetry tab to Default values doesnt help...

Error Message:
Warning
Simulator for this Firmware is not yet available

before changing this Simulation worked.

kilrah, 2015-10-30:
No way a change in settings could have caused it.

It could however come form a companion update, or something on your PC. Check in companion's install folder that you see files named opentx-simulator-*.dll.

You could also just uninstall and reinstall companion.

trashmaster9000, 2015-10-30:
All Simulator.dll's are here.
I uninstalled Companion, downloaded it and Installed it again.

I think the file gets Corrupted when i save it.
It only gets corrupted if i Change Settings in the telemetry Screen.

I made a Video:
https://www.youtube.com/watch?v=_RfKIY258Xk

Here is the test File from the Video:
https://onedrive.live.com/redir?resid=B4766C2F15754EC7!2757&authkey=!AGrBrQgYmmTPK5c&ithint=file%2cbin

projectkk2glider, 2015-10-30:
Must be a Windows only thing, because I can not reproduce on 2.1.3 or in current master. 

@trashmaster9000 what happens if you open Companion normally and then load the bad eeprom from the Companion itself?

projectkk2glider, 2015-10-30:
Idea, perhaps current working dir is changed by such start up and then the dll's are not found!

projectkk2glider, 2015-10-30:
:arrow_up:  Yes that is it. On windows the simulator dll files are only searched in "./" dir, which obviously is the dir in which the eepe or bin file resides. This is only the case when the companion is started by double clicking on eepe or bin file.

In such case (on linux) we then search in predefined hard-coded path where the simulator libs are installed by the installation package.

On Windows that part is missing. We could just determine the Companion.exe location and search there.

trashmaster9000, 2015-10-30:
if i start Companion and load the File Simulator works...

projectkk2glider, 2015-10-30:
Yes that proves it then, thanks for reporting. We will prepare a fix, but in the meantime you can use the work-around.

projectkk2glider, 2015-10-30:
@kilrah would you be able to test my fix? I really dislike running Windows VM...

closed 2015-10-31 by kilrah

kilrah, 2015-10-31:
I fixed the fix :)

jtaylor2, 2015-10-31:
This change broke the Cygwin (posix)  compile on Windows.  I'll submit a pull request to fix it later today..

jtaylor2, 2015-10-31:
I have created a pull request that removes  __CYGWIN__ from the Issue 3012 Windows pathing fix (it broke building Companion and Simulator with Cygwin on Windows). While Cygwin creates .exe and .dll files on Windows it uses posix pathing and not Windows pathing.  It also defines __GNUC__ as well as __CYGWIN__.

projectkk2glider, 2015-10-31:
Thanks to both!

commit 22c429a1 (2015-10-31): Also search for simulators in companion.exe path on Windows (Fix #3012)