opentx/opentx #3015: 2.1 Lua playNumber() not performing automatic conversions like 2.0

dsbeach, 2015-10-30:
The tables below show observed behavior of the Lua playNumber() function:

I'm assuming the differences in units are intentional and will reflect them in the new Lua documentation.  However, should the automatic conversions for distance, speed, and temperature be incorporated in 2.1? 

If so, are these the correct new values for automatically converted items?

|Desc. | 2.0 value | 2.1 value | Notes |
| --- | --- | --- | --- |
|speed| 5 | 7 | input in KPH |
|distance|6|9| input in meters|
|temperature|7|11| input in Celsius|
|speed|14|?|input in Knots|

|	2.0 Unit	|	Sound	|	File (.wav)	|	Automatic conversion rules	|
|	---	|	---	|	---	|	---	|
|	0	|		|		|		|
|	1	|	Volts	|	116	|		|
|	2	|	Amps	|	118	|		|
|	3	|	Meters per Second	|	120	|		|
|	4	|	*missing file*	|	122	|		|
|	5	|	Kilometers per Hour / Miles per Hour	|	124 / 142	|	Input value is KPH	|
|	6	|	Meters / Feet	|	126 / 140	|	Input value is meters	|
|	7	|	Degrees	|	128	|	Input value is celsius, converted to Fahrenheit for Imperial	|
|	8	|	Percent	|	130	|		|
|	9	|	Milliamps	|	132	|		|
|	10	|	Milliamp Hours	|	134	|		|
|	11	|	Watts	|	136	|		|
|	12	|	DB	|	138	|		|
|	13	|	Feet	|	140	|		|
|	14	|	Kilometers per Hour / Miles per Hour	|	124 / 142	|	Input value is in Knots, converted to KPH or MPH	|
|	15	|	Hours	|	144	|		|
|	16	|	Minutes	|	146	|		|
|	17	|	Seconds	|	148	|		|
|	18	|	RPM	|	150	|		|
|	19	|	Gee	|	152	|		|
|	20	|	Degrees	|	128	|		|
|	21	|	*missing file*	|	156	|		|
|	22	|	*missing file*	|	158	|		|
|	23	|	Point Zero	|	160	|		|
|	24	|	Point Two	|	162	|		|
|	25	|	Point Four	|	164	|		|
|	26	|	Point Six	|	166	|		|
|	27	|	Point Eight	|	168	|		|
								
								
|	2.1 Unit	|	Sound	|	Sound File (.wav)	|		
|	---	|	---	|	---	|		
|	1	|	Volts	|	116	|		
|	2	|	Amps	|	118	|		
|	3	|	Milliamps	|	120	|		
|	4	|	Knots	|	122	|		
|	5	|	Meters per Second	|	124	|		
|	6	|	Feet per Second	|	126	|		
|	7	|	Kilometers per Hour	|	128	|		
|	8	|	Miles per Hour	|	130	|		
|	9	|	Meters	|	132	|		
|	10	|	Feet	|	134	|		
|	11	|	Degrees Celsius	|	136	|		
|	12	|	Degrees Fahrenheit	|	138	|		
|	13	|	Percent	|	140	|		
|	14	|	Milliamp Hours	|	142	|		
|	15	|	Watts	|	144	|		
|	16	|	DB	|	146	|		
|	17	|	RPM	|	148	|		
|	18	|	Gee	|	150	|		
|	19	|	Degrees	|	152	|		
|	20	|	Milliliters	|	154	|		
|	21	|	Fluid Ounces	|	156	|		
|	22	|	Hours	|	158	|		
|	23	|	Minutes	|	160	|		
|	24	|	Seconds	|	162	|		
|	25	|	Point Four	|	164	|		
|	26	|	Point One	|	166	|		
|	27	|	Point Three	|	168	|		
|	28	|	Point Five	|	170	|		
|	29	|	Point Seven	|	172	|		
|	30	|	Point Nine	|	174	|

kilrah, 2015-10-30:
There are no automatic conversions on 2.1, the unit that is played is the one you specify from this enum.
https://github.com/opentx/opentx/blob/master/radio/src/myeeprom.h#L1148

dsbeach, 2015-10-30:
I'll update the documentation and close the issue.

closed 2015-10-30 by dsbeach