opentx/opentx #3017: [Bug ??][Radio [ Companion 2.1.3] Toggle On/Off the internal RF is not seen by Companion

LapinFou, 2015-10-30:
Hi folks,

Step to reproduce the problem:
1- create a new model on the radio (Internal RF is ON / D16 mode).
2- Load on Companion. At his point everything is working as expected.
3- Turn Off the internal RF under Companion and write back to the radio.
4- On the radio turn ON the RF and chose D16 - Ch:1-16 - No impulse 
5- Load on Companion, the internal RF is still OFF ?!
:smiley:

projectkk2glider, 2015-10-30:
If the point 4 is true, then the setting is not taking effect on the radio and Companion is not to blame. 

But on the other hand I find it really hard to believe that RF module setting on the radio is not working. We would have a lot of unhappy users. 

Something does not make sense here. Can you double check your description.

LapinFou, 2015-10-30:
I checked The 4th point: the radio does take in account the internal RF On, but Companion doesn't see it.

closed 2015-10-31 by kilrah

kilrah, 2015-10-31:
@bsongis This is actually what was causing the issue I reported multiple times and that you were dismissing as "old eeprom from before the Taranis internal PPM was implemented"!

commit 3d29b8e7 (2015-10-31): Missing default for internal module type (Fix #3017)