opentx/opentx #3020: Firmware Build Option timer3 and Companion Model Setup Timer 3 Settings

elecpower, 2015-11-01:
It appears there is no check and warning in Companion when you define a third timer for a model and the Build Options timer3 is unchecked in Companion Settings.
Have confirmed on Taranis X9D Plus with bootloader 2.1.3 with Companion 2.1.3 on Win 10 that the model is written to the radio with the 3rd timer settings by reading the radio settings back and they match. 
The radio does no display the 3rd timer as one would expect due to the firmware build option for timer3.
Suggest user is warned at some point of the inconsistency in their settings. Maybe at the time of writing the model to the radio.

kilrah, 2015-11-01:
Already solved by hiding timer3 settings if option not selected.

closed 2015-11-01 by kilrah

elecpower, 2015-11-01:
Radio side is okay but Companion display does not match the firmware.

kilrah, 2015-11-01:
I know it doesn't in 2.1.3, but it has already been fixed for next release.

elecpower, 2015-11-01:
okay. Thanks for the advice.