opentx/opentx #3023: Splash Screen Wrong Color Depth Bricks Taranis X9D Plus

elecpower, 2015-11-02:
Bootloader and Companion 2.1.3
After bricking required disconnecting battery to enable 3 button power on to launch bootloader to repair.

I realise this was my fault however to prevent others making the same mistake could Companion check the splash screen image file when the user selects and before writing with the firmware to the radio to ensure image propoerties match current supported format.

This feature could be beneficial in future versions if the supported format is changed especially if an old version format was not updated by the user as part of them upgrading.

kilrah, 2015-11-02:
Companion automatically converts the images to the appropriate format so I don'e see how that could happen. Unless there's a bug in that section.

elecpower, 2015-11-02:
Unless something went wrong in the firmware write process. It said it was successful but maybe an error was not trapped. On normal radio power up the screen was totally blank. Also blank when exiting Bootloader mode.

Initially after detecting the issue the radio was not detected (no STMBOOTLOADER device) so maybe a Win 10 fail in the middle of the write. I initially thought the battery voltage had fallen too low so recharged and after a few hours the radio was detected and able to write with 4 bit image.

Big image properties 212px W X 64px H, 8 bits, BMP, 14.3KB.

Would you like me to retry with the big image file to test if this was just a Win 10 glitch. If it fails then I have a brick again but can recover again fingers crossed or would you like to do a quick check of the code?

Please advise.

kilrah, 2015-11-02:
Bad writes can happen indeed.
Which flashing method are you using? Power off, power on form companion, or from SD card? The first and last should be rock-solid but the 2nd is dependent on the way the computer OS decides to do the write.

kilrah, 2015-11-02:
Just tested with a few random images including a 28MB, 10MP full color BMP and companion loads and converts it without issue, the resulting firmware works fine. 

So I'd say you most likely had a write error.

closed 2015-11-02 by kilrah

elecpower, 2015-11-02:
Flash method. Power Off. Connected USB. Checked Devices for STM. Started Companion and downloaded 2.1.3 noheli lua en and wrote to radio with incorrect 8 bit splash.

kilrah, 2015-11-02:
Hmm. Please zip the offending BMP and upload it to some file sharing site e.g. wetreansfer, then post the link here so we can have a look at that particular file.

elecpower, 2015-11-02:
Link to file http://we.tl/UfMaUZC64M

kilrah, 2015-11-02:
OK, works fine here both just selecting that file in the flashing dialog, and making a new bin with it with the "Edit radio splash image" command.

Can't see an issue. Please detail every step you do if you can reproduce consistently.

elecpower, 2015-11-02:
Looking like it was a random OS driver issue. I will try to reproduce and advise the steps if I can.
Appreciate your prompt testing and responses.